qmk/qmk_toolbox #6: when flashing with avrdude device does not reset

giammin, 2017-11-27:
i tried to flash my atmega 32u4 but qmktoolbox stuck when lunching avrdude.

no message or other. i just see the command line of avrdude and nothing happens

i tried to put the device in reset mode but qmktoolbox did not find the correct com port when the device is in flashing mode.

i see from device manager (windows 10) that atmega 32u4 when connect has a com port lets say per example com4.

when  the device goes in flashing mode and during the 5 second timespan it has a different com port. and it is always -1 respect the normal com port it is assigned when connected (so if com4 is assigned, during reset it is com3)

then i  put the device in flashing mode and lunched directly avrdude (from a command line) with the com -1 port and i was able to flash qmk.

when searching online a solution i found that you can force the atmega to reset connecting at 1200bps  on com port.

i think qmktoolbox when falshing with avrdude should use the  1200bps open/close trick then get the rigth com port and finally  run avrdude on that port

drashna, 2017-11-27:
Is this a SparkFun board? (eg, the SparkFun Pro Micro)?  
If so, then it's Device ID isn't rocognised yet, but should be added in the next release.

Otherwise, it may be a good idea to find out what the COM port's device ID is (when it's in bootloader mode).

giammin, 2017-11-27:
it is a cheap promicro copy (OPEN-SMART Atmega32U4).

I'll check for the device ID tonight at home

giammin, 2017-11-27:
VID: 2341
PID: 8037

jackhumbert, 2018-02-04:
We have support for this device now - thanks!

closed 2018-02-04 by jackhumbert