qmk/qmk_toolbox #10: Support Adafruit Feather BLE 32u4

colemarkham, 2017-12-13:
The Feather BLE 32u4 uses the Catalina bootloader, but uses 239a:000c for the USB device ID.

colemarkham, 2017-12-13:
Tried changing the USB device ID in USB.m, but when I run from XCode it's not working, even for the Pro Micro. I assume it's not getting device events, but I'm not sure.

drashna, 2017-12-13:
Which OS? 
For windows, you'd add that here:
https://github.com/qmk/qmk_toolbox/blob/a0bdb18340893c46edc5106a9f81b55e46171975/windows/QMK%20Toolbox/USB.cs#L61
For OS X, here:
https://github.com/qmk/qmk_toolbox/blob/198002b3d1d6d82fe9176bb496471026e728ed8e/osx/qmk_toolbox/USB.m#L102
And probably here: https://github.com/qmk/qmk_toolbox/blob/198002b3d1d6d82fe9176bb496471026e728ed8e/osx/qmk_toolbox/Flashing.h#L17

colemarkham, 2017-12-13:
I'm on OSX. The problem is that it just doesn't seem to be working at all when I run from XCode.

But I just realized that I had my USB cable plugged into another computer. Now I feel dumb. Still need to put the new device ID in, but I should be able to make the changes now.

drashna, 2017-12-13:
That would do it!  Been there, done that! :)

closed 2017-12-14 by jackhumbert