qmk/qmk_toolbox #13: Is it supposed to work on macOS High Sierra ?

danielo515, 2017-12-27:
Hello,

First of all, thanks for this tool, I'm using it on windows and it's awesome.
I just tried the mac version on macOS High Sierra ( 10.13.2 ) and it does not even open. Do you have any report about this ? I tried your latest released version.

Thanks in advance.

danielo515, 2017-12-27:
I just generated an error report, hope this helps:

```
Process:               QMK Toolbox [56301]
Path:                  /private/var/folders/*/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
Identifier:            fm.qmk.toolbox
Version:               0.0.8 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QMK Toolbox [56301]
User ID:               501

Date/Time:             2017-12-27 08:56:29.697 +0100
OS Version:            Mac OS X 10.13.2 (17C88)
Report Version:        12
Anonymous UUID:        CA7FEDEA-8B00-343B-FF8E-62EB2E9E70B1

Sleep/Wake UUID:       9EBE0B55-438D-4B15-8644-C2368F582E93

Time Awake Since Boot: 250000 seconds
Time Since Wake:       740 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/lib/libusb-1.0.0.dylib
  Referenced from: /private/var/folders/*/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
  Reason: image not found

Binary Images:
       0x10189e000 -        0x1018a7ff7 +fm.qmk.toolbox (0.0.8 - 1) <0B202E30-616A-3248-9C06-E9BC8FEB1833> /var/folders/*/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
       0x1052fd000 -        0x10534798f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld
    0x7fff39c83000 -     0x7fff39d1efff  com.apple.framework.IOKit (2.0.2 - 1445.31.1) <7556086A-EFAF-3262-9822-6F666CFC6655> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit

Model: MacBookPro9,2, BootROM MBP91.00D7.B00, 2 processors, Intel Core i5, 2,5 GHz, 16 GB, SMC 2.2f44
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x04CD, 0x46412D313630304331312D38475351000000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x04CD, 0x46412D313630304331312D38475351000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF5), Broadcom BCM43xx 1.0 (7.21.190.16.1a2)
Bluetooth: Version 6.0.2f2, 3 services, 18 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: CDC Composite Gadget, Ethernet, en4
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS545050A7E362, 500,11 GB
Serial ATA Device: HL-DT-ST DVDRW  GS41N
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1
```

danielo515, 2017-12-27:
Ok, this single command execution fixed my problem:

```bash
brew install libusb
```

Maybe you want to add it to some kind of FAQ.

Regards

danielo515, 2017-12-27:
Ok, new problem.
Now the software starts and it recognizes the connected device, but when it tries to load the following error appears:

````
>>> avrdude -p atmega32u4 -c avr109 -U flash:w:/Users/danielo/myProgrammingProjects/qmk_firmware/danielomp_default.hex:i -P /dev/cu.usbmodem1421 -C avrdude.conf
    dyld: Library not loaded: /usr/local/lib/libusb-0.1.4.dylib
      Referenced from: /var/folders/jf/d7sk0z691c53n7c7l6qcz5qr0000gn/T/AppTranslocation/38B2E794-B9BD-4F12-9AB0-13CACE12B36F/d/QMK Toolbox.app/Contents/Resources/avrdude
      Reason: image not found
```

May it be a problem because I have installed libusb using brew ?

therg, 2017-12-29:
I had a similar issue on OSX 10.11. After I installed avrdude with Homebrew I was able to flash firmware.
```
brew install avrdude
```

TurboMech, 2018-01-22:
@jackhumbert I just had the same issue with another person.  Using Revision 0.0.9 causes issues for some users on Mac OS High Sierra not installing the libusb properly.  I cannot confirm if Revision 0.0.8 had the same issue as I had all the build tools installed previously to using QMK Toolbox Revision 0.0.8.

iansinnott, 2018-01-28:
Ran into this issue as well on High Sierra. As noted above `/usr/local/lib/libusb` was not found. Did a brew install (`brew install libusb`) but to no avail. The issue was that I had homebrew installed at `$HOME/homebrew` rather than the standard `/usr/local`. Creating a new homebrew installation at `/usr/local` and _then_ running `brew install libusb` did the trick. Now the toolbox starts up as expected.

Perhaps the toolbox is assuming that `brew --prefix` will be `/usr/local`?

jackhumbert, 2018-02-04:
Yeah, this should be included and linked with the executable - not entirely sure why this isn't happening quite yet. Placing the .dylib files from the repo in the same folder as the executable may solve this issues.

luandy64, 2018-02-26:
Hi all,

I installed the Build tools as directed in the docs and actually was flashing QMK for a while. I had to restore my machine from a Time Machine back up though and found that QMK Toolbox wasn't working. I ran into this same issue and got the same error report as @danielo515 in the comment up there. My fix went as follows:

1. `brew doctor` Reveled `/usr/local/lib` was not writable, so I ran the command it suggested
2. `sudo chown -R $(whoami) /usr/local/lib`

Then I was back in business.

kmanwar89, 2018-07-09:
Just wanted to add my two cents.  I just built a macro pad that uses a ProMicro controller, and has QMK firmware.  I'm on a 2017 MacBook Pro running High Sierra 10.13.5.  This is a vanilla system that was just reinstalled last week, and today was my first time using the graphical flashing tool.  I encountered the _libusb_ error and resolved that with a simple **brew install libusb**.  Once QMK Toolbox actually opened, I encountered the _image not found_ error as well, and resolved this with a **brew install avrdude**.   Everything worked without issue after that.

leico, 2018-08-27:
macbook Pro 2016
High Sierra 10.13.5
XCode 9.4.1
vitamins included

Give up. I tried to run this app. 
Create script to compile required binaries, libraries from homebrew
<https://github.com/leico/qmk_toolbox/blob/master/osx/replace.sh>

and rebuild app with xcode. It was success and running QMK Toolbox on High Sierra.
but **default hex ---It was downloaded by Toolbox, can't write, died usb port** . Recover from ICSP and avrdude I needed.
Other side build on my mac hex is working.

Below is build binary but 
**Attention: Firmware write fails and may need to be restored from the ICSP port of AVR, or become a brick your keyboard if no ICSP port on your PCB. I recovered 3 times vitamins each hands.**
It is so experimental build. 

[QMK Toolbox.app.zip](https://github.com/qmk/qmk_toolbox/files/2323050/QMK.Toolbox.app.zip)

kergoth, 2018-08-30:
This should really be built to avoid hardcoding paths on disk like /usr/local/lib, and should really ship the libraries it depends upon. Worked around it after the fact here with:
```
$ mkdir -p ~/Downloads/QMX\ Toolbox.app/Contents/lib/
$ cp $HOMEBREW_PREFIX/Cellar/libusb/*/lib/libusb-1.0.0.dylib ~/Downloads/QMK\ Toolbox.app/Contents/lib/
$ install_name_tool -change /usr/local/lib/libusb-1.0.0.dylib @rpath/libusb-1.0.0.dylib ~/Downloads/QMK\ Toolbox.app/Contents/MacOS/QMK\ Toolbox
$ install_name_tool -add_rpath @loader_path/../lib ~/Downloads/QMK\ Toolbox.app/Contents/MacOS/QMK\ Toolbox 
```

gretel, 2019-06-07:
had to `brew install libusb-compat libftdi`, too.

FancyChancie, 2019-06-20:
> had to `brew install libusb-compat libftdi`, too.

This fixed it for me as well.

fauxpark, 2019-06-20:
@FancyChancie @gretel does it still work if you uninstall those packages? I can run the Toolbox just fine without them, but I did have `libusb` and `libusb-compat` installed before (and perhaps `libftdi` at some point too).

Also, did you install the Toolbox cask, or download it directly?

fauxpark, 2019-06-21:
Right, I did some digging and came to the conclusion that this particular issue is resolved.

Outside of my Xcode build folder and the Toolbox repo, I could not find any instances of these dylibs after uninstalling the brew formulae. The Toolbox continued to work.

Versions 0.0.9 and 0.0.10 do not seem to have a dependency on libusb:
```
$ otool -L QMK\ Toolbox
QMK Toolbox:
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1570.15.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1671.40.118)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1570.15.0)
```

Version 0.0.8, however, does:
```
$ otool -L QMK\ Toolbox 
QMK Toolbox:
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/usr/local/lib/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1252.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1225.1.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1404.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1253.0.0)
```

It seems this is thanks to [this commit](https://github.com/qmk/qmk_toolbox/commit/db47da66535d6d8c66773c9cf33a1cca645ccb70#diff-63d0c9d392919c1376b8848ea8974867) where the libusb dylib was moved out of Frameworks and turned into a simple CopyFiles build step into the .app's Resources directory. If I remove those two dylibs and their references in the project file, I can no longer flash, but the Toolbox still starts.

However, flashing with avrdude is still not possible unless you install `libftdi`, so we should be bundling that as well.

closed 2019-07-13 by drashna

bsberry, 2019-08-23:
Leaving this here if it helps anyone else: prompted by @luandy64's [comment earlier in the thread](https://github.com/qmk/qmk_toolbox/issues/13#issuecomment-368366583) I did a `brew doctor` and saw 

```
Warning: Unbrewed .la files were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected .la files:
  /usr/local/lib/libusb-1.0.la
```

So I did

```
rm /usr/local/lib/libusb-1.0.la
brew reinstall libusb
```

and everything started working again.

Hope that's helpful.

PR #95 (2019-07-13): Fix libusb and add libftdi to get avrdude flashing without homebrew formulae

## Description

Trying to flash with avrdude throws a missing dylib error unless you install avrdude (more specifically, libftdi0) through Homebrew. So we need to bundle it here along with libusb and libusb-compat. While I was at it, I replaced the whole lot with the dylib files from these Homebrew bottles:

https://homebrew.bintray.com/bottles/libusb-1.0.22.mojave.bottle.tar.gz
https://homebrew.bintray.com/bottles/libusb-compat-0.1.5_1.mojave.bottle.tar.gz
https://homebrew.bintray.com/bottles/libftdi0-0.20.mojave.bo [... cut]