qmk/qmk_toolbox #17: Bad command

Bennoo, 2018-01-27:
Hello,

I just installed qmk-toolbox and I would like to flash PROMICRO based keyboard,

But after the boot of the application:
```
*** QMK Toolbox (http://qmk.fm/toolbox)
    Supporting following bootloaders:
     - DFU (Atmel, LUFA) via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via teensy_loader_cli (https://pjrc.com/teensy/loader_cli.html)
     - STM32 (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - Kiibohd (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
    And the following ISP flasher protocols:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
*** Caterina device connected: Périphérique d’entrée USB -- ::
*** Caterina device connected: Périphérique série USB (COM3) -- ::
*** Caterina device connected: Périphérique USB composite -- 2341:8037:6
```

I select my .hex file and hit the Flash button and I get this:
```
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"C:\Users\ben\Downloads\iris_rev2_default_5ee3e304.hex":i -P 
    C:\Users\ben\AppData\Local\QMK\QMK Toolbox\0.0.9\avrdude.exe: option requires an argument -- P
    Usage: avrdude.exe [options]
    Options:
      -p <partno>                Required. Specify AVR device.
      -b <baudrate>              Override RS-232 baud rate.
      -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
      -C <config-file>           Specify location of configuration file.
      -c <programmer>            Specify programmer type.
      -D                         Disable auto erase for flash memory
      -i <delay>                 ISP Clock Delay [in microseconds]
      -P <port>                  Specify connection port.
      -F                         Override invalid signature check.
      -e                         Perform a chip erase.
      -O                         Perform RC oscillator calibration (see AVR053). 
      -U <memtype>:r|w|v:<filename>[:format]
                                 Memory operation specification.
                                 Multiple -U options are allowed, each request
                                 is performed in the order specified.
      -n                         Do not write anything to the device.
      -V                         Do not verify.
      -u                         Disable safemode, default when running from a script.
      -s                         Silent safemode operation, will not ask you if
                                 fuses should be changed back.
      -t                         Enter terminal mode.
      -E <exitspec>[,<exitspec>] List programmer exit specifications.
      -x <extended_param>        Pass <extended_param> to programmer.
      -y                         Count # erase cycles in EEPROM.
      -Y <number>                Initialize erase cycle # in EEPROM.
      -v                         Verbose output. -v -v for more.
      -q                         Quell progress output. -q -q for less.
      -l logfile                 Use logfile rather than stderr for diagnostics.
      -?                         Display this usage.
    
    avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>

```
It seems the avrdude command is bad formatted.. Do I need to upgrade soemthing or this is an issue?

Regards,

jackhumbert, 2018-02-04:
It looks like it wasn't able to find the port from the USB string - the regex might be making some assumptions that aren't accurate for different languages.

mavanmanen, 2018-02-05:
Currently investigating this issue.

closed 2018-02-06 by jackhumbert

Cereal916, 2019-06-09:
I am having this same exact issue.

PR #22 (2018-02-06): Bugfix/17 bad command
Using a more reliable way to detect a device's com port by using `Select * from Win32_SerialPort` and comparing the usb device's DeviceID to the com port's PNPDeviceID.

Should fix #17