qmk/qmk_toolbox #18: Autoflash is dangerous

drashna, 2018-02-03:
Since this is happened REPEATEDLY to me....  (You'd think I'd learn, but here we are)...

Would it be possible to "lock down" the autoflash to  single device (as an option), or enable "only if mod/shift/key is held down" option?
This would prevent idiots like me from flashing the wrong firmware on their boards (and potentially bricking the controller)

skullydazed, 2018-02-04:
Just thinking aloud here, what if Autoflash turned off after a timeout? Say 1 hour after the last flash?

drashna, 2018-02-04:
That would work too.

mavanmanen, 2018-02-04:
Maybe turn off if there hasn't been a flash for more than like 10 minutes?

jackhumbert, 2018-02-04:
It might not hurt to have a separate "Flash when ready" checkbox that deactivates once it flashes, like a one-shot.

mavanmanen, 2018-02-06:
Should be closed, fixed in #24

closed 2018-02-06 by jackhumbert