qmk/qmk_toolbox #32: Currently available latest release reports itself as 0.0.8, github says the latest version is 0.0.9

skullydazed, 2018-03-04:


jackhumbert, 2018-03-29:
Is this with the installer, or the standalone app? I might have messed up the release.

skullydazed, 2018-03-29:
This is in the about box.

kalupa, 2018-04-21:
@jackhumbert It's installing 0.0.8 both the app.zip and the .pkg version

rjventrone, 2018-05-08:
@jackhumbert brew also install the 0.0.8 version

techsock, 2018-07-22:
I'm seeing the same on macOS with both installer and standalone.

fauxpark, 2018-08-16:
The Windows build reports the correct version. On macOS, brew downloads the latest release from GitHub tagged 0.0.9, but the about box shows 0.0.8 - this is taken from Info.plist, and it was incremented in https://github.com/qmk/qmk_toolbox/commit/984a263ae7ae2753bb33abe8601171d108731d0c, just minutes before the 0.0.9 release was tagged. However it appears it did not make it into the build.

fauxpark, 2018-09-13:
@jackhumbert: I see you've fixed the 0.0.9 release, the .app now displays the correct version, so this issue should be resolved. However, you changed the filename from `QMK.Toolbox.app.zip` to `QMK_Toolbox.app.zip` and now Brew can't install it :( Would you mind fixing this, or telling me which one is correct? Cheers.

jackhumbert, 2018-09-13:
@fauxpark Good catch! Just renamed it :)

closed 2018-09-13 by jackhumbert