qmk/qmk_toolbox #33: Mention drivers for dfu-util

333fred, 2018-03-14:
Not sure if here is the right place, or if it would be better in the ergodox infinity readme, but it would be good to mention that boards that need to be flashed with DFU will likely need the WinUSB driver for flashing to be successful, otherwise you'll see this message:
```
dfu-util: No DFU capable USB device available.
```
I used Zadig (http://zadig.akeo.ie/) to install the driver after I remembered I needed it, but I lost a good 30 minutes to attempting to troubleshoot the issue then finding the driver.

skullydazed, 2018-04-02:
Funny enough I opened #35 just now before noticing this issue. We have qmk_driver_installer which our custom fork of Zadig for installing those drivers automatically. For the moment users can use that to install drivers: https://github.com/qmk/qmk_driver_installer/releases

Longer term QMK Toolbox should handle this automatically. Leaving this issue open to track that.

gwerbin, 2019-01-08:
Just dropping in to note that this is was issue for me on the Pro Micro as well, so it's probably affecting more people than might be immediately apparent.

drashna, 2019-10-01:
I believe that #122 should resolve this.

closed 2019-10-09 by drashna