qmk/qmk_toolbox #34: Is Linux version just a draft?

ulgens, 2018-03-19:
I tried to compile and run `linux` folder but i see nothing but empty screen with buttons.

[image]

Also, the folder itself looks like has no functional code except UI.

skullydazed, 2018-04-02:
Sorry about the delay responding here, currently Linux is not supported. That's some GUI framework code that @jackhumbert has been playing around with in order to support Linux in the future.

ulgens, 2018-04-02:
Thanks :disappointed:

closed 2018-04-02 by ulgens

mollat, 2018-09-11:
Please write into the readme - so nobody else wastes time on it...

paines, 2020-05-17:
Please, reopen this ticket. The linux version is just a dumb placeholder. The buttons do nothing at all.