qmk/qmk_toolbox #36: RESET and Reset are confusing

skullydazed, 2018-04-02:
When flashing a keyboard you often press a physical RESET button on the PCB, or you use the `RESET` key code, which exists keyboard mode and enters dfu mode. Inside the QMK Toolbox there's a button labeled Reset whose job it is to exit flashing mode and enter keyboard mode. More than a few new users are getting confused when they press Reset in the toolbox and their keyboard doesn't enter flashing mode.

I'm not sure what the best solution here is, but the first possibility that comes to mind is changing "Reset" to "Exit DFU".

fauxpark, 2019-03-28:
Perhaps the button could start out disabled and only become usable when a bootloader is detected.
Relabelling it to "Exit DFU" would drive the point home even more.

drashna, 2019-12-30:
Yeah, I think "Exit DFU" or "Exit Bootloader" may be clearer.

skullydazed, 2020-05-16:
This seems to be done now. Thanks @nooges!

closed 2020-05-16 by skullydazed