qmk/qmk_toolbox #43: Request for new official release

tkchris93, 2018-06-02:
I am running into an issue on the latest official release that was later fixed in #22. I don't know how to build the code from source on Windows, so I would really appreciate a new official release. Thanks!

kavu, 2018-07-01:
@jackhumbert Hi, Jack! Can you give an estimate on this? Thanks!

takashiski, 2018-10-09:
@jackhumbert We cannot load even default keymaps from qmk.fm by toolbox, but I think it resolve #38. We need new release ASAP.

amatecha, 2018-10-22:
On that note, the Mac download link is no longer working (as of this moment).  Guess I'll try v0.0.8…

kevinduffey, 2018-11-18:
Doesnt seem the link works @coarse? Just gives me empty page. I am still having the same Toolbox issue that was shown last year.. e.g. plugging in an IRIS keyboard controller (trying to flash them) doesnt work. I am hoping I followed the chatter on that issue that got me here and that this is resolved/fixed and hopefully a new toolbox for windows is available to use?

drashna, 2018-11-29:
I've also updated the exe in the 0.9 release with the most recent version.

Durnlaw, 2018-12-05:
I am having the same error from [#17](https://github.com/qmk/qmk_toolbox/issues/17). Which I think is in line with what others are saying on this thread? Sorry, new to this.

dmcgits, 2019-03-30:
I'm looking for the BootloadHID support too :D 
Weirdly there's BootloadHID support in Drashna's exe, I can load the jj50 firmware, but I can't see any way to connect to a v-usb device

drashna, 2019-05-18:
Well, I'm working on this. I finally figured out how to make the installer, actually.  

Only issue is that I don't have any Apple devices, and no way to compile it for MacOS :(

drashna, 2019-05-29:
New release added.  Grab 0.0.10

closed 2019-05-29 by drashna