qmk/qmk_toolbox #47: Malware detected in mcufont.exe of release 0.0.9

onimoger, 2018-07-16:
Bitdefender deleted qmk_toolbox.exe claiming it contained a ransom malware:

`qmk_firmware\lib\ugfx\tools\mcufontencoder\binaries\windows\mcufont.exe`
`SHA256 Hash: 727952580c14ecb0f39408aa3ffe568b5486c55dd5b91f9e86a167263e289a8b`
`Malware Name: Trojan.GenericKD.30594631`
`Threat Type: Malware, Status:Deleted`

Maybe it's a false alarm, but it should be looked into.

jackhumbert, 2018-07-16:
I've had issues with for a couple years - I think it's just a false alarm. If anyone has any idea what's going on with it, or an alternative version that doesn't raise this, I'd be happy to change things!

sgmpunisher, 2018-07-17:
You might want to run Malwarebytes and figure out which files have been infected. BitDefender is known to bypass it. However, blinding removing the files may render some of the popular programs unusable. Backup and running Malwarebytes in Safe Mode and using Avast or other latest AV should help.

closed 2018-08-15 by jackhumbert