qmk/qmk_toolbox #52: Need a way to automate QMK Toolbox releases

drashna, 2018-10-08:
#51 adds this for OS X, but we need to add the ability for both OSX and Windows

jackhumbert, 2018-10-10:
@nooges do you have any experience automating either of these via Travis?

nooges, 2018-10-10:
It's been a while since I've done anything with Travis.

drashna, 2018-11-29:
AppVeyor may be a better way to do this, and supports GitHub Releases. 

https://www.appveyor.com/docs/deployment/github/

Though it doesn't appear to support MacOS images, so that may be an issue to compile for.
https://www.appveyor.com/docs/build-environment/

drashna, 2019-10-01:
Well, I've gotten this down to a science, for the manual process, including updating brew.   So that's a plus. 

But it would be nice to have this fully automated.

StrikerRUS, 2019-10-04:
[Azure Pipelines](https://docs.microsoft.com/en-us/azure/devops/pipelines/?view=azure-devops)  CI service is completely free for open source projects and has a task named [`GitHubRelease`](https://docs.microsoft.com/en-us/azure/devops/pipelines/tasks/utility/github-release?view=azure-devops), which can publish any artifacts as a release at GitHub (even it can optionally create a changelog based on the commits after a previous release). It has Linux, macOS and Windows support ([different versions](https://docs.microsoft.com/en-us/azure/devops/pipelines/agents/hosted?view=azure-devops)). In addition, there is a Docker Hub built-in support, if you need fully customized environment.

drashna, 2020-06-11:
Well, this has been added, and automated. No need to build locally, aside from testing/development now.

closed 2020-06-11 by drashna