qmk/qmk_toolbox #56: Reset EEPROM doesn't work for atmega32u2

colemarkham, 2018-10-31:
Clicking the "Reset EEPROM" runs `dfu-programmer atmega32u2 erase --force` with a message that prints 'success'. There is no indication that the EEPROM was not actually reset. To support atmega32u2, the empty EEPROM file just needs to be added to the build alongside the 32u4 file, but any other processor that doesn't work should print a message.

fauxpark, 2019-03-28:
It seems like `atmega32u4_eeprom_reset.hex` only erases the first 128 bytes. Evidently QMK currently doesn't use more than that.

Considering all supported MCUs have far more than 128 bytes of EEPROM, I don't think we need an extra hex file - the existing one can probably serve as a generic EEPROM clearer for AVRs.

drashna, 2019-05-18:
@fauxpark all we need to do is wipe the first byte of EEPROM, actually.  If that's cleared, it's not `FEED` (aka the magic number for eeprom), and will trigger an initialization on the next startup of the firmware.  

Honestly, I think it would be better to include an eep file, and flash that instead. That would be more "compatible" anyways.

fauxpark, 2019-05-19:
Something like this then:

```
:0100000000FF
:00000001FF
```

PR #75 (2019-05-20): Use a single generic eep file for resetting EEPROM

## Description

Removed the MCU-specific `atmega32u4_eeprom_reset.hex` for a more universal `.eep` file which sets the first byte to `0x00` so that the first two bytes are no longer `0xFEED` (which should trigger EEPROM reinit).

Works on macOS with a DFU keyboard. Windows, and Caterina on macOS still need testing.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #56
closed 2019-05-20 by drashna