qmk/qmk_toolbox #59: Linux ever?

TrinityErik, 2019-01-02:
Any plans to finish the coding for Linux? Anyone working on it or is it dead in the water?

BramVer, 2019-02-10:
It seems pretty dead in the water, which is a shame since I've dropped my Windows machine, and none of the ways of 'flashing on Linux' seem to work for me.  
I basically have to spin up a Windows VM for the sole purpose of flashing compiled .hex files onto controllers.  

Having the qmk_toolbox run on Linux would be greatly appreciated.

sanecito, 2019-04-16:
I'm willing to throw a bounty on this via BountySource. Would need to know the desired amount though from @jackhumbert or someone reputable so I'm not just tossing money at something that gets locked up and nothing actually happens.

I too hate having to spin up a Windows VM or work laptop, and even then Windows drivers tend to be a bit buggy when flashing stuff it seems.

QuercusFelis, 2019-10-01:
Any update on this? I can compile the linux code but the program itself is still unusable

drashna, 2019-12-30:
Unfortunately, the linux version is basically just a shell.  Unless somebody wants to work on the linux version, there won't be any progress.

AliceGrey, 2020-02-20:
@sanecito I'd also be willing to chip in on BountySource if @jackhumbert or @fauxpark took this up.

jackhumbert, 2020-02-20:
I do have a renewed interest in this now that early Pinephones are out there - getting something going on Ubuntu Touch would be pretty neat.

fauxpark, 2020-02-21:
I'm not at all familiar with developing on Linux, much less a GUI application. From what I can tell, a Linux port would be kinda distro-specific, unless we used something like Electron? (I have no idea)

tzarc, 2020-02-21:
Perhaps something like:
https://github.com/zserge/webview/tree/webview-x

sanecito, 2020-02-21:
There shouldn't be a need to target only a specific distro. Without deep diving, odds are one could repurpose a decent amount of the C# code and to the Mono framework. Looking at the Linux dir, if they wanted to continue along a C/C++ route for logic they could use either the GTK or Qt toolkits for GUI. Those two especially with recent advancements like libhandy for GTK are toolkits that are mo polished and better lend themselves to adaptive UI which is needed to say, make a solid UX for apps on PinePhone, Librem5.

rkrhlikar, 2020-03-01:
I recently created a [fork](https://github.com/rkrhlikar/qmk_toolbox/tree/linux-support) where I took the Glade layout file and started implementing some of the logic behind it using gtkmm (the C++ gtk library), so it should work on different distros. I started with the low hanging fruit (meaning not touching anything related to hardware, which is, of course, the most important part) so as not to lose all motivation right out of the gate (working with HW is always a PITA :)). I expect to test flashing Pro Micros sometime this week.
As is often the case with these things, I make no promises about actually seeing this through. I decided to comment here, so that even if I do run out of steam, anyone interested in picking this up can do so without starting from scratch. As long as I do, however, keep working on it, I will be commenting with any significant progress I make in the near future (unless that would be considered spamming?).

rkrhlikar, 2020-03-08:
As promised, a quick update:
I made some progress on the hardware side of things. I was able to successfully flash firmware onto the Pro Micros on my Kyria keyboard using the GUI. That means that the current test implementation for Caterina devices is up and running.

I had some problems with avrdude on my computer which were resolved by stopping the ModemManager service on my Ubuntu 18.04 installation when flashing the Arduinos. This can be done by running `sudo systemctl stop ModemManager.service`. I found this fix on the [Arduino forums](https://forum.arduino.cc/index.php?topic=580326.0) (if anyone would like to get more details considering the issue).

Planned next steps:
- Support other devices that are programmed using avrdude (should be relatively straight-forward as the only difference is in the parameters passed to avrdude; however, I won't be able to properly test as I don't have any devices to test with)
- Test implementation for dfu devices via dfu-programmer (I can test that on my GH60)
- Add libraries that are used by the program (gtkmm, curlpp, ...) so that it can be distributed as an archive and ran directly after extraction (no need for dependency installation on the user system)
- Implement everything required to run qmk toolbox directly from the terminal (currently the only way to flash is through the GUI)

Long(er)-term goals:
- Support the rest of the devices (same as with avrdude devices, I will have some trouble testing as I don't own any compatible keyboards, so it will probably require some community involvement :))
- Implement HID console
- Update readme
- Code cleanup

Erovia, 2020-03-19:
@rkrhlikar if you need some testing, ping me on Discord!

NoraCodes, 2020-10-07:
@rkrhlikar please let me know if you need any additional help with this. I'd love to see a QMK Toolbox working on Linux and I have some experience distributing GTK applications to different distros which could potentially be useful.

CharString, 2020-11-05:
This is comment is just as a reference for some Linux user contemplating their first build and this issue might scare them.

I found the CLI experience of the firmware just really good. I installed toolbox, because it was step one in the manual that came with my pcb. But I ended up not using it. Somewhere along the way I switched to following [the qmk newbie docs](https://docs.qmk.fm/#/newbs_getting_started) and used the CLI for everything. `qmk setup` even detected I was on Fedora at one time and Debian the other, and used their respective package managers to install the missing dependencies. For the first flash, `qmk doctor` and `dmesg` helped with seeing whether my board was in bootloader mode or not.

There's just one thing I did differently. As I do a lot of python development myself, I chose to use [pipx](https://pipxproject.github.io/pipx/) as I do with all my python packages I just want to install for the CLI command they provide.

My guess to why no one has picked this issue up as something to work on, is that it doesn't really scratch a personal itch. That said, Vim is my editor of choice, Super+Space fires up a terminal window in my setup and I do have experience with Arduino. So your idea of a good experience might be different from mine. But as you're building your own keyboard... using CLI instead of GUI just gives you more use and gratification from your project.

jberryman, 2021-11-12:
Has anyone managed to get it to run in wine? I've struck out there

malikbenkirane, 2022-05-18:
@CharString your comment saved my life that should be merged to README.md :+1:

brunzefb, 2022-12-27:
[Linux UI for QMK](http://blog.brunzema.com/wp-content/uploads/2022/12/QMK_LinuxUI.png)

brunzefb, 2022-12-27:
2022-12-26-Update 2: I have made some progress with the Linux [QMK Toolbox main UI](http://blog.brunzema.com/wp-content/uploads/2022/12/QMK_LinuxUI.png), coding this using the Avalonia (WPF-like) framework, just because I worked with WPF a few years ago.  It all came back pretty quickly.

brunzefb, 2022-12-28:
2022-12-28 Update: More progress to report.  The main UI mostly works now, here is a [new screenshot](http://blog.brunzema.com/wp-content/uploads/2022/12/qmk_progress.png). The back-end USB connection detection now works (with some things still not right).

brunzefb, 2022-12-30:
2022-12-30: Update: More progress.  The WinForms original UI and the [Avalonia UI look almost identical now](http://blog.brunzema.com/wp-content/uploads/2022/12/winavalonia.png). I have added menus, shortcuts, and reworked the UI to make it very similar to the Windows version. Drag and drop files works on Windows, but not on Linux, although the code is identical.

brunzefb, 2022-12-30:
2022-12-30 Update: The LibUsbDotNet was not giving me the names of the devices, given a PID/VID. Reason for this is that  one must call Open() on the device, which gets more info, including the Vendor name.  Now Open only works if you are root :(, but you can add an [entry into /etc/udev/rules.d](https://stackoverflow.com/questions/22713834/libusb-cannot-open-usb-device-permission-isse-netbeans-ubuntu), and reload with udevadm, but this also needs sudo :-(. But you can get info as non-root by running udevadm info /dev/bus/usb/[device], which has VID, PID, and names.  This can be parsed... find /dev/bus/usb -name "*" -exec udevadm info {} \;    will list all the info...

plovinicius, 2024-07-13:
Any update on this?

brunzefb, 2024-07-13:
Three does not seem to be much appetite for the project to consider my PR,
and the code works on my Linux box...

fauxpark, 2024-07-13:
https://github.com/qmk/qmk_toolbox/pull/396#issuecomment-2136711691

closed 2024-07-13 by fauxpark

sallymanda, 2025-03-26:
seriously? i can understand there not being resources to add linux support at the present time. but taking active steps to block it?

tzarc, 2025-03-26:
You're more than welcome to describe and demonstrate concrete solutions to address the issues raised.

sallymanda, 2025-03-26:
my concern is not whether i am welcome to suggest solutions; i can do so with or without being welcomed to. my concern regards the "not planned" status of this issue which, along with #396, indicates such suggestions will not be engaged with.

hramrach, 2025-03-26:
Flashing tools are in no way special. If the application itself can run on arbitrary distribution (and there are packaging methods like AppImage that do just that) so can the flashing tools.

The problem is getting access to the device. On Windows you need to install a device driver, on Linux you need to install an udev rule that makes the device accessible to users.

Not all that different as far as I can tell.

fauxpark, 2025-03-26:
They are special. We don't build the flashing tools, we just scrape them from the packages, and they are generally not statically linked. So a `dfu-programmer`, for example, taken from the corresponding package from a glibc-based distro will probably not work on a musl-based distro.

tzarc, 2025-03-26:
Of course not. What’s special about the utilities are the dependencies, most notably things like the version of glibc they’re linked against. Newer distros will provide newer copies and thus compiling and linking against those copies results in incompatible binaries when considering older distributions.

This is compounded by the fact that other dependencies get bumped; in one case `libtinfo.so.5` was one of the dependencies of one of the tools, but one of my distros provides `libtinfo.so.6` which is incompatible with binaries linked against `.5`. Should toolbox provide multiple binaries of the same tool, detecting at runtime which one to use? Which combinations of glibc and tinfo (given the example) should be built/provided? Add libusb 1.0, libusb 0.1, libusb-compat, libserialport, libhidapi and libftdi into the mix; which combinations should be linked against? How many different copies of `avrdude` do we end up with? `dfu-programmer`, `dfu-util`, `mdloader`, `teensy_loader_cli`, and `wb32-dfu-updater_cli` need similar treatment. 

Are you prepared to do the legwork to find which combinations work with each distribution, going back a few versions in each case?

hramrach, 2025-03-26:
Not building them but scraping them is your choice. It's not like the tools manifest out of thin air for other platforms, either.

And how is building the tools more of a supply problem than building the application itself?

tzarc, 2025-03-26:
> And how is building the tools more of a supply problem than building the application itself?

Refer to the lack of bandwidth earlier in the discussion chain.

Ultimately, this isn't an endeavour that the team is prepared to take on given other focus; you're free to fork QMK Toolbox and address all the concerns, and if/when you've got them solved we can look at potentially pulling such functionality across. We aren't looking at accepting incomplete solutions that only work for a subset of distributions, because invariably the QMK team would be the ones expected to provide the support and diagnostics for getting things working on (insert user's favourite distribution here).

As it stands, given the current support of the QMK CLI and the relative simplicity of `qmk setup`, you're going to have to forgive us for viewing this as a very low priority work item.

Hasshu, 2025-03-26:
Could this be solved with Flatpak?

tzarc, 2025-03-26:
Perhaps, someone needs to work it out.

This discussion has reached its conclusion as far as the QMK team is concerned; there are enough potential paths discussed but motivated individuals are going to need to pursue them. Locking this issue as there's little value in receiving future complaints with minimal action.

drashna, 2025-03-29:
> Could this be solved with Flatpak?

No.

https://docs.flatpak.org/en/latest/sandbox-permissions.html

* No access to any host files except the runtime, the app, ~/.var/app/$FLATPAK_ID, and $XDG_RUNTIME_DIR/app/$FLATPAK_ID. Only the latter two being writable.
* No access to the network.
* ***No access to any device nodes (apart from /dev/null, etc).***
* No access to processes outside the sandbox.
* Limited syscalls. For instance, apps can’t use nonstandard network socket types or ptrace other processes.
* Limited access to the session D-Bus instance - an app can only own its own name on the bus.
* No access to host services like X11, system D-Bus, or PulseAudio.

The "no access to any device nodes bit really being the main issue. 
IIRC, you can allow specific devices through, but that requires explicitly listing each device, individually.  Which basically makes toolbox pointless.