qmk/qmk_toolbox #60: Error flashing pro micro on windows

Micro-Biology, 2019-01-04:
Issue occurs whilst using qmk toolbox on windows 8.1 enterprise edition.

Possibly related to [this issue #42](https://github.com/qmk/qmk_toolbox/issues/42).

`avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"<hex-file-location>\macro12_default.hex":i -P 
    <qmk-install-location>\QMK\QMK Toolbox\0.0.9\avrdude.exe: option requires an argument -- P
    Usage: avrdude.exe [options]`

I assume it its just a error and it should be:
`avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"<hex-file-location>\macro12_default.hex":i -P "<qmk-install-location>\QMK\QMK Toolbox\0.0.9\avrdude.exe: option requires an argument"`

gwerbin, 2019-02-10:
Same error, looks like some things got mixed up in the most recent release.

Micro-Biology, 2019-02-11:
Does anyone know how to use an older version? (or a new version with this fixed?)

Telchii, 2019-02-24:
I was experiencing the same issue tonight. After making sure QMK Toolbox was up to date, I did some googling and found an [/r/MechanicalKeyboards thread](https://www.reddit.com/r/MechanicalKeyboards/comments/9tzh89/helpflashing_pro_micro/) from November 2018, with the following recommendations:

1. Install the Arduino IDE software package: https://www.arduino.cc/en/Main/Software (Try this one first and make sure you're not trying to start a flash after the device disconnects.)

2. Run the qmk_driver_installer. Download the zip, extract it and run the "install_all_drivers" script. https://github.com/qmk/qmk_driver_installer/releases

After installing both of these, I was able to flash my pro micro. If you have auto-flash enabled, make sure it's set before you plug in any USB devices. I did have short the reset (RST) and ground (GND) pins a couple times (due to the small window to start the flash and auto flash not picking up my pro micro) and manually click Flash before the device disconnected.

gwerbin, 2019-03-05:
@Micro-Biology download it from the Releases page in this repository.

BramVer, 2019-03-07:
@drashna Updated the .exe in the latest release as said [here](https://github.com/qmk/qmk_toolbox/issues/43#issuecomment-442972731)

Micro-Biology, 2019-03-13:
> I've also updated the exe in the 0.9 release with the most recent version.

Unfortunately I'm already on this release, I did try reinstalling, still has the bug.
I think its just missing some quotations for the install location, my coding is very limited (especially on windows) or I'd attempt to fix it myself.

BramVer, 2019-03-13:
The same problem here, couldn't get any release to work anymore, both on a win 8.1 VM or a win 8.1 dedicated machine.  
I know a buddy with a Macbook and that one works fine though.

drashna, 2019-05-20:
on a newly downloaded version, does this still happen? 
(eg, redownload 0.0.9)

kncarrier28, 2019-05-25:
I just downloaded and get the same error. It's saying that the -P flag is given with no arguments. This flag is supposed to have the serial port for the device, so I'd guess that issue is with pulling in that serial port.

drashna, 2019-10-01:
This may be fixed now?

whitehot6er, 2019-11-24:
I just downloaded v 0.0.14 on Win 10 yesterday (11/22/19) and encountered the same problem. Using version 0.0.13 I was able to successfully flash my pro micro.

yanfali, 2019-11-24:
> I just downloaded v 0.0.14 on Win 10 yesterday (11/22/19) and encountered the same problem. Using version 0.0.13 I was able to successfully flash my pro micro.

If you've upgraded windows recently it may have uninstalled the drivers. They do this on every major upgrade. Try reinstalling using zadig.

fauxpark, 2019-11-24:
Zadig won't help, Caterina is driverless.

I don't see any changes between 0.0.13 and 0.0.14 that would affect the behaviour of avrdude (barely anything on the Windows side of things was touched). It is probably just a coincidence that downgrading worked - this particular error seems to be somewhat of a race condition related to the appearance of the COM port and the USB device providing it.

fauxpark, 2019-11-24:
So as it turns out, there *was* a change that affected avrdude (more specifically, the COM port detection) after all. My bad. The PR linked above will fix this; for now just use the previous version.

closed 2019-11-29 by fauxpark

louia, 2021-04-27:
Not fixed for me, it just doesnt work with pro micro and QMK Toolbox 0.0.21 and avrdude 6..3

fauxpark, 2021-04-27:
Could you post a screenshot?

louia, 2021-04-27:
Yes, absolutly. My pro Micro is new, its the first time with QMK.

[image]

fauxpark, 2021-04-27:
Have a look in Device Manager (hit `View->Devices by container`, and `Show hidden devices`). Uninstall any devices under the COM12 device, and tick the "uninstall driver" checkbox if there is one. Then try again.

marcoklein, 2021-05-19:
Got the exact same error. Same versions as @louia - works if i execute the command manually and append `com7` in my case for the port:

`avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"C:\Users\Marco\Documents\GitHub\qmk_firmware\crkbd_rev1_legacy_default.hex":i -P com7`

(Note the com7 after the -P)

[image]

Error:
[image]

St3v3n-4n4, 2022-07-27:
se ha solucionado el error?

PR #135 (2019-11-29): Send ManagementBaseObject instance to GetComPort()

## Description

This was due to me changing what information `deviceId` held in #129, and not realising this gets passed to `GetComPort()` 🤦‍♂ Since it no longer matched up with the `PNPDeviceId` associated with the COM port, nothing would be found. Instead let's just give `GetComPort()` the entire object so it can extract whatever it needs.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #60