qmk/qmk_toolbox #61: .NET error on startup

gwerbin, 2019-01-08:
Using the latest version of QMK Toolbox on Windows I get this [error](https://imgur.com/G8lZato):

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at QMK_Toolbox.Usb.DetectBootloader(ManagementBaseObject instance, Boolean connected)
   at QMK_Toolbox.Usb.DetectBootloaderFromCollection(ManagementObjectCollection collection, Boolean connected)
   at QMK_Toolbox.MainWindow.MainWindow_Load(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at QMK_Toolbox.MainWindow.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
```

I seem to be able to proceed fine by pressing the "Continue" button in the error pop-up.

I also get this error when I press the "Say Hello" button:

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   at QMK_Toolbox.MainWindow.sayHelloButton_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
qmk_toolbox
    Assembly Version: 0.0.9.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Users/GMWERB/Downloads/qmk_toolbox.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
HidLibrary
    Assembly Version: 3.2.46.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Users/GMWERB/Downloads/qmk_toolbox.exe
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Deployment
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Deployment/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Deployment.dll
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.IO.Compression.FileSystem
    Assembly Version: 4.0.0.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Users/GMWERB/Downloads/qmk_toolbox.exe
----------------------------------------
Newtonsoft.Json
    Assembly Version: 10.0.0.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Users/GMWERB/Downloads/qmk_toolbox.exe
----------------------------------------
Syroot.Windows.IO.KnownFolders
    Assembly Version: 1.0.2.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Users/GMWERB/Downloads/qmk_toolbox.exe
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

ClarkWillBattle, 2019-01-21:
So do I.

tdogg40, 2019-01-28:
Getting the same as you guys. I am able to press continue and then flash my board with no issues though.
Yesterday I was using the toolbox with no issue at all but today I did a fresh install of windows on my PC and that's when I started getting the error.

jmanj78, 2019-05-12:
Also getting this on Windows 10. Same thing, able to progress to the toolbox if I hit Continue. I am then unable to detect my board (bm16s), though as this is the first time I've tried using QMK, that could be user error and unrelated.

RailiO, 2019-05-17:
Same issue

stoppablemurph, 2019-08-19:
> I think I fixed it... It is when your driver does not install properly

How did you fix it? I understand the problem, but I'm new to the software and the whole keyboard flashing deal in general.. <.<

RailiO, 2019-08-19:
I can currently not home so if you can wait about 4 days I’ll help you out

drashna, 2019-08-19:
> How did you fix it? I understand the problem, but I'm new to the software and the whole keyboard flashing deal in general.. <.<

Basically, use Zadig to install the correct driver.  From there, it should stop crashing, in theory. 

https://docs.qmk.fm/#/faq_build?id=unknown-device-for-dfu-bootloader

trashcandrummer, 2019-09-15:
Can confirm. Had the same problem and installing the driver again with Zadig fixed it

drashna, 2019-10-01:
It this still happening in the 0.0.13 version?

Duckle29, 2019-10-25:
yea :/

fauxpark, 2019-10-27:
I had a look at this today, and the error seems to be caused by the following line in `DetectBootloader()`:

```cs
var uni = instance.GetPropertyValue("ClassGuid").ToString();
```

Evidently, when there is no driver installed, the `GetPropertyValue()` call returns null, and so the `ToString()` throws the exception. IMO, the GUID is not very useful (and there is no equivalent on macOS), and it happens that I already removed it in #129.

The next release should basically eliminate this crash as it will include that PR and #128, so on first run you will be prompted to install drivers.

closed 2019-11-04 by drashna

AwakeJJ, 2023-12-24:
im still getting this error "class  not registered" ive tried everything and nothing is working for me

fauxpark, 2023-12-24:
That's not the same error as in the OP. Please open a new issue with more information.

PR #129 (2019-11-04): Bootloader detection improvements

## Description

The third number shown in the "device connected/disconnected" messages should be the device revision - this should be a more reliable method of grabbing that. I'm hoping this will allow us to detect QMK-DFU vs Atmel DFU and not have to do a full chip erase to clear the EEPROM, by setting the revision to something like the Unicode codepoint for Ψ (03A8). Also some various tweaks to clean up my work tree. :)

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Clos [... cut]