qmk/qmk_toolbox #62: Error popup trying to download a default layout

gwerbin, 2019-01-23:
1. Open QMK Toolbox
2. Select Tokyo60 under "keyboard from qmk.fm" (or almost any other keyboard...)
3. Click "Load"

Error:

<detail>

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Net.WebException: The request was aborted: Could not create SSL/TLS secure channel.
   at System.Net.WebClient.DownloadFile(Uri address, String fileName)
   at System.Net.WebClient.DownloadFile(String address, String fileName)
   at QMK_Toolbox.Form1.setFilePath(String filepath)
   at QMK_Toolbox.Form1.loadKeymap_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
qmk_toolbox
    Assembly Version: 0.0.9.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Program%20Files%20(x86)/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3314.0 built by: NET472REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
HidLibrary
    Assembly Version: 3.2.46.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Program%20Files%20(x86)/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Deployment
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Deployment/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Deployment.dll
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.7.3221.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Newtonsoft.Json
    Assembly Version: 10.0.0.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Program%20Files%20(x86)/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
System.Numerics
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Numerics/v4.0_4.0.0.0__b77a5c561934e089/System.Numerics.dll
----------------------------------------
System.Runtime.Serialization
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Runtime.Serialization/v4.0_4.0.0.0__b77a5c561934e089/System.Runtime.Serialization.dll
----------------------------------------
System.Data
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3260.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_32/System.Data/v4.0_4.0.0.0__b77a5c561934e089/System.Data.dll
----------------------------------------
System.IO.Compression.FileSystem
    Assembly Version: 4.0.0.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Program%20Files%20(x86)/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------
Syroot.Windows.IO.KnownFolders
    Assembly Version: 1.0.2.0
    Win32 Version: 0.0.9
    CodeBase: file:///C:/Program%20Files%20(x86)/QMK%20Toolbox/qmk_toolbox.exe
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

</detail>

fauxpark, 2019-06-14:
Most likely fixed in 0.0.10 (https://github.com/qmk/qmk_toolbox/commit/ba1e5d4f0acd5818a2c37cd70460453e29e1617a) - if so this can be closed.

elfmimi, 2019-06-19:
I tried the above. Tokyo60 is now OK.
Yet ARM based boards are still NG.
for example, planck/rev6 tries to download from this URL.
https://qmk.fm/compiled/planck_rev6_default.hex
while the correct URL must be with .bin extension.
https://qmk.fm/compiled/planck_rev6_default.bin

fauxpark, 2019-06-19:
Yeah, it's currently hardcoded to ask for the .hex file. That's unrelated to this particular issue, though.

PR #133 (2019-11-29): Precompiled firmware download: try .bin if .hex not found

## Description

Replaces the `.hex` extension with `.bin` if the attempt to download the former failed. Bit of a hack, but it works 👍

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #126
* Closes #62
closed 2019-11-29 by fauxpark