qmk/qmk_toolbox #63: Block .hex uploads when using dfu-util

pelrun, 2019-02-02:
Dfu-util doesn't support ihex formatted files, but it will happily open them and send them as raw data... which obviously doesn't work.

Have Toolbox detect such attempts and stop with an error explaining that the .bin version needs to be used instead.

drashna, 2019-02-13:
Well, that should be fixed on the back end now.

Specifically, we should be pushing BIN files to qmk.fm, which is where the toolbox grabs the firmware images from. 
Now, we just need to figure out how to grab them, when appropriate.

pelrun, 2019-02-13:
Fixing the backend ensures that the downloader doesn't provide a bogus file, but it doesn't help when people use Toolbox to flash their own binaries, or ones they downloaded manually. Which is what happens for a large proportion of people we've been helping on Discord with this problem.

Toolbox shouldn't be letting you flash garbage to a kb (at least, when it's easily detected.) Ideally the dfu-util devs would pull their heads out and add that check themselves, but I don't see that happening any time soon.

PR #66 (2019-05-14): Check for .bin extension before attempting flash with dfu-util
Related issue(s) (if any):
Fixes #63.

Changes:
Added a check to see if the filename ends with ".bin" before trying to use dfu-util.
I can't test this as I don't have an STM32 keyboard, but the logic is pretty simple and I'm reasonably confident it does the right thing. Testers appreciated, of course.
closed 2019-05-14 by jackhumbert