qmk/qmk_toolbox #70: Are the Servers down?

Fluffy347, 2019-05-04:
OS:Win10

OS Version:Latest

QMK Toolbox version:0.09

Issue description:Are the servers down? QMK Toolbox just installed and it says "trying to get the keyboard list from QMK.FM, you might not have a internet connection or the servers are down."

QMK Toolbox log:*** QMK Toolbox (https://qmk.fm/toolbox)
    Supporting following bootloaders:
     - DFU (Atmel, LUFA) via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via teensy_loader_cli (https://pjrc.com/teensy/loader_cli.html)
     - STM32 (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - Kiibohd (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - BootloadHID (Atmel, ps2avrGB, CA66) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
     - Atmel SAM-BA via mdloader (https://github.com/patrickmt/mdloader)
    And the following ISP flasher protocols:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
    Something went wrong when trying to get the keyboard list from QMK.FM, you might not have a internet connection or the servers are down.

I indeed do have an internet connection however. And When I hit the "Say Hello" button it gives me an error saying it cannot do this.

I mean it happens if the server(s) is(are) down, but I just haven't seen anyone else with this issue.

fauxpark, 2019-05-05:
See #69

Fluffy347, 2019-05-05:
Okay, my understanding of that link is that a line calling to a URL should be changed, but i'm not quite sure how/where I should be looking for a file to change that link. Have I missed something in that link?

fauxpark, 2019-05-05:
I was just pointing you at the PR that fixes your issue, it hasn't been merged yet.

If you don't need to download precompiled .hex files directly from the Toolbox (the "Keyboard from qmk.fm" section), you can ignore the error message. Flashing firmware files you've downloaded from the Configurator or compiled yourself will still work.

Fluffy347, 2019-05-06:
Okay cool. I have a .HEX all set and compiled. It's just the toolbox application to flash the board that I need. I'll be fine for the time being however. I'll just wait for the merge to happen.

Thanks!

davisthegreate, 2019-05-16:
older version that i downloaded 10/13/2018 does not have this issue, although it is the same version number

fauxpark, 2019-05-16:
It won't show the error message, but it will probably still fail to download the keyboard list.

drashna, 2019-05-29:
Fixed now.

closed 2019-05-29 by drashna