qmk/qmk_toolbox #74: Error opening HID Manager

a7hybnj2, 2019-05-16:
## Describe the Bug
```
*** QMK Toolbox (http://qmk.fm/toolbox)
    Supporting following bootloaders:
     - DFU (Atmel, LUFA) via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via teensy_loader_cli (https://pjrc.com/teensy/loader_cli.html)
     - STM32 (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
     - Kiibohd (ARM) via dfu-util (http://dfu-util.sourceforge.net/)
    And the following ISP flasher protocols:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
*** Halfkay device connected
*** Attempting to flash, please don't remove device
>>> teensy_loader_cli -mmcu=atmega32u4 /Users/user1/Downloads/lfkeyboards_lfkpad_ayout_numpad_6x4_mine-2.hex -v
    Teensy Loader, Command Line, Version 2.1
    Read "/Users/user1/Downloads/lfkeyboards_lfkpad_ayout_numpad_6x4_mine-2.hex": 21460 bytes, 66.5% usage
    Error opening HID Manager
    Unable to open device
```

## System Information

 - Operating system: macOS Mojave - 10.14.4
 - QMK Toolbox version: 0.0.9 (1)

## Additional Context
All the dependencies were installed via brew and are on most recent version.  
I originally installed via cask but got the error so I uninstalled via cask and manually downloaded the file but received the same error.

I looked up the error and saw something about ATOM but nothing with a clean solution or a something I would know how to implement.

a7hybnj2, 2019-05-17:
So, I just fixed this issue...

I found this: https://github.com/alswl/teensy_loader_cli/commit/9c16bb0add3ba847df5509328ad6bd5bc09d9ecd

I cloned the repo `make` and then took the resulting teensy_loader_cli and replaced the one in the `qmk toolbox.app' and it worked!

a7hybnj2, 2019-05-17:
hahah I did just get this error when trying to use the reset button:
```
*** Halfkay device connected
>>> teensy_loader_cli -mmcu=atmega32u4 -bv
    Teensy Loader, Command Line, Version 2.1
    Error opening HID Manager you stupid bananaface
    Found HalfKay Bootloader
    Booting
*** Halfkay device disconnected
```
I don't feel like a "stupid banannaface" but I like it...

lbibass, 2019-09-09:
> hahah I did just get this error when trying to use the reset button:
>
> ```
> *** Halfkay device connected
>     Teensy Loader, Command Line, Version 2.1
>     Error opening HID Manager you stupid [...]

Hey, sorry for necro-ing this, but how did you replace the existing teensy loader CLI and replace the one in the qmk toolbox app?

a7hybnj2, 2019-09-09:
@lbibass I listed the instructions above but in the sake of making it more clear:

Download this https://github.com/alswl/teensy_loader_cli and extract it somewhere (or just clone it)
 - I don't think there has been a PR on from this repo but I haven't used the most recent version of QMK Toolbox.

In terminal CD to the extracted folder and 'make' (this will take a min or two).

Take the teensy_loader_cli that was just compiled and use it to replace the one in the "qmk toolbox.app" that you installed earlier. Navigate to "qmk toolbox.app" and right click - show contents(in case you didn't know how to do that bit).

### again I haven't used the latest version of QMK Toolbox but this is how I worked around it 5 months ago

lbibass, 2019-09-09:
Oh, I know how to show contents, and I have compiled it already, but do you really just drag the new program into QMK Toolbox? That seems a bit too easy, haha.

fauxpark, 2019-09-09:
The current version (0.0.12) uses the latest binaries taken from Homebrew bottles, targeted at High Sierra, so they should work on 10.13 and up. Is it still doing the same thing?

a7hybnj2, 2019-09-09:
@fauxpark thanks for the version update. I was running 0.0.9 when I had the issues that my above explanation fixed. When I need to reprogram my keyb I will be sure to try the latest version.

drashna, 2019-10-01:
Any update on this?  We're at 0.0.13 now, so hopefully this has been resolved.

lbibass, 2019-11-10:
> Any update on this? We're at 0.0.13 now, so hopefully this has been resolved.

I can confirm that it is still broken. It only works with the fix that @a7hybnj2 so helpfully provided.

fauxpark, 2019-11-10:
Hm, so the LFKPad doesn't use a Teensy; the chip is onboard. I set the bootloader in its rules.mk to `atmel-dfu` recently, based on this fact - but perhaps it is running a HID bootloader instead (which might explain why the Toolbox is detecting it as a Teensy).

I know that LUFA has an equivalent of HalfKay, but IIRC they are not compatible with each other.
@lbibass do you also have an LFKPad? I'm interested to know how it shows up in System Information.app.

lbibass, 2019-11-10:
I don't, sorry. All I know is that I cannot flash a teensy nor a pro-micro without some hacks, and the pro micro flashing procedure seems to be entirely broken on Catalina.

fauxpark, 2019-11-10:
Does flashing work using the Teensy Loader GUI, at least? https://www.pjrc.com/teensy/loader_mac.html

lbibass, 2019-11-10:
Oh, I wish! The reason I had to move to using QMK toolbox in the first place was that the Teensy Loader GUI is 32 bit. So it doesn't launch on Catalina. 

The fix that was provided seems very easy, why not just use that?

fauxpark, 2019-11-10:
Seems to launch fine for me, though I don't have a Teensy to test if it actually flashes.
[image]

lbibass, 2019-11-10:
Well, this is entirely new to me! They must have updated the app since I last tried to use it. Thanks for letting me know it works!

fauxpark, 2019-11-10:
As for the provided fix, generally I don't really want to be compiling these flasher binaries just for the Toolbox. They are lifted from the precompiled Homebrew bottles, which I hope have more people testing/using, so we can point at them or upstream when something goes wrong (and they likely already know about it). In this case, I think the best course of action would be to file an issue over on the teensy_loader_ci repo so that it is actually properly fixed. The thread linked in alswl's commit seemed to go nowhere.

ELLIOTTCABLE, 2020-04-30:
Still broken, mentioned fix still works. `(=`