qmk/qmk_toolbox #77: Add ability to delete entries in file list for MacOS side

drashna, 2019-05-20:
## Feature Request Type

- [ ] Core functionality
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

This feature does/will exist for the Windows side, but not the macOS side. So to keep feature parity, it would need to be added on the MacOS side.

fauxpark, 2019-05-21:
It's seeming non-trivial to add a keyDown event handler to the combobox in a similar way to how #68 does it, so I'm still in favour of having a proper Open Recent menu for both.

fauxpark, 2019-05-21:
Well, macOS doesn't "do" hover selections for combobox dropdowns, apparently. But otherwise I've got it working.

fauxpark, 2019-05-21:
The combo box is populated through the recent documents list managed by NSDocumentController, which doesn't provide any way to remove specific entries from the list...so this wouldn't be persistent.

I *could* get it to send `clearRecentDocuments:` when the number of items reaches zero...but...egh.

nooges, 2020-03-23:
Random thought that's a hack: call `clearRecentDocuments:` then repopulate with `noteNewRecentDocumentURL:` with deleted item excluded.

PR #355 (2022-04-28): [macOS] Fix open menu item and recent documents clear behaviour

## Description

- `File->Open Recent->Clear` did not immediately clear the file path combobox
- `File->Open` threw an exception because the `openDocument:` method was not implemented - redirected it to `openButtonClick:`

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #77
  Marking this as fixed because I think the "clear all" approach is better and simpler. Will need to be redone on Windows - there is no inbuilt func [... cut]
closed 2022-04-28 by fauxpark