qmk/qmk_toolbox #84: QMK Toolbox Crash when flashing

airyimbin, 2019-05-30:
I am using a STM32 board, with QMK Toolbox version 0.0.10 on MacOS 10.14.5.
I did "make projectkb/alice:default" to create the bin. When I hit "Flash" on QMK Toolbox it crashes immediately and the window disappears. No apparent error messages are shown.

This was working fine on v0.0.9.

airyimbin, 2019-05-30:
Actually here is the error log I got from it.

QMK Toolbox quit unexpectedly.
```
Process:               QMK Toolbox [97988]
Path:                  /Applications/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
Identifier:            fm.qmk.toolbox
Version:               0.0.10 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QMK Toolbox [97988]
User ID:               502

Date/Time:             2019-05-30 16:20:22.633 -0400
OS Version:            Mac OS X 10.14.5 (18F132)
Report Version:        12
Anonymous UUID:        045AECBE-4814-7BDF-1D3F-BE10F75C53D6

Time Awake Since Boot: 870000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'launch path not accessible'
abort() called
terminating with uncaught exception of type NSException

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff354a5cfd __exceptionPreprocess + 256
1   libobjc.A.dylib                     0x00007fff5fb4fa17 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff354a5b2f +[NSException raise:format:] + 201
3   Foundation                          0x00007fff3769881f -[NSConcreteTask launchWithDictionary:error:] + 1291
4   QMK Toolbox                         0x00000001047be85f -[Flashing runProcess:withArgs:] + 569
5   QMK Toolbox                         0x00000001047bf65f -[Flashing flashSTM32WithFile:] + 336
6   QMK Toolbox                         0x00000001047be9eb -[Flashing flash:withFile:] + 234
7   QMK Toolbox                         0x00000001047c0503 __32-[AppDelegate flashButtonClick:]_block_invoke + 132
8   libdispatch.dylib                   0x00007fff612d063d _dispatch_client_callout + 8
9   libdispatch.dylib                   0x00007fff612d2de6 _dispatch_continuation_pop + 414
10  libdispatch.dylib                   0x00007fff612e1f42 _dispatch_source_invoke + 2056
11  libdispatch.dylib                   0x00007fff612db54b _dispatch_main_queue_callback_4CF + 813
12  CoreFoundation                      0x00007fff353efcd7 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
13  CoreFoundation                      0x00007fff353ef401 __CFRunLoopRun + 2289
14  CoreFoundation                      0x00007fff353ee8be CFRunLoopRunSpecific + 455
15  HIToolbox                           0x00007fff346da96b RunCurrentEventLoopInMode + 292
16  HIToolbox                           0x00007fff346da6a5 ReceiveNextEventCommon + 603
17  HIToolbox                           0x00007fff346da436 _BlockUntilNextEventMatchingListInModeWithFilter + 64
18  AppKit                              0x00007fff32a74987 _DPSNextEvent + 965
19  AppKit                              0x00007fff32a7371f -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
20  AppKit                              0x00007fff32a6d83c -[NSApplication run] + 699
21  AppKit                              0x00007fff32a5cd7c NSApplicationMain + 777
22  libdyld.dylib                       0x00007fff6131d3d5 start + 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff614582c6 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff61513bf1 pthread_kill + 284
2   libsystem_c.dylib             	0x00007fff613c26a6 abort + 127
3   libc++abi.dylib               	0x00007fff5e59e641 abort_message + 231
4   libc++abi.dylib               	0x00007fff5e59e7df default_terminate_handler() + 267
5   libobjc.A.dylib               	0x00007fff5fb51ee3 _objc_terminate() + 97
6   libc++abi.dylib               	0x00007fff5e5aa19e std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff5e5aa213 std::terminate() + 51
8   libdispatch.dylib             	0x00007fff612d0651 _dispatch_client_callout + 28
9   libdispatch.dylib             	0x00007fff612d2de6 _dispatch_continuation_pop + 414
10  libdispatch.dylib             	0x00007fff612e1f42 _dispatch_source_invoke + 2056
[... 467 lines cut ...]
__FONT_DATA                          4K        1 
__LINKEDIT                       222.7M        9 
__TEXT                           286.2M      374 
__UNICODE                          564K        1 
mapped file                       94.7M       44 
shared memory                     3036K       13 
===========                     =======  ======= 
TOTAL                              1.3G     1031 
TOTAL, minus reserved VM space   949.5M     1031 

Model: MacBookPro12,1, BootROM 184.0.0.0.0, 2 processors, Intel Core i5, 2.7 GHz, 8 GB, SMC 2.28f7
Graphics: kHW_IntelIris6100Item, Intel Iris Graphics 6100, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445364533303445452D45474346000000
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445364533303445452D45474346000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.12f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
USB Device: STM32  BOOTLOADER
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5
```

drashna, 2019-05-31:
Could you try reinstalling/redownloading, and see if that helps?

airyimbin, 2019-05-31:
I used AppCleaner to uninstall just now and redownloaded the pkg version to install again.
I still had the same issue.

```
Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'launch path not accessible'
abort() called
terminating with uncaught exception of type NSException
```

airyimbin, 2019-05-31:
I also uninstalled and and used the zip version and had the same issue.

fauxpark, 2019-05-31:
This was due to dfu-util not having the executable bit set in Git, which propagated to the final build artifact. We should not be relying on Git permissions; instead we need to add a build script to the Xcode project that ensures all the flasher binaries are executable before packaging it up. See #86.

As a workaround for this, just run:
```
chmod +x /Applications/QMK\ Toolbox.app/Contents/Resources/dfu-util
```
and flashing should work again.

airyimbin, 2019-05-31:
Great! It's working now!

closed 2019-05-31 by airyimbin

tobico, 2019-06-12:
I also encountered this crash. Would have saved me some time I spent preparing a bug report if this issue had been left open, since it's not yet fixed in the current release.

sizemo, 2019-10-15:
> This was due to dfu-util not having the executable bit set in Git, which propagated to the final build artifact. We should not be relying on Git permissions; instead we need to add a build script to [...]

This isn't working for me. I get no response when I put that in Terminal and when I try to flash afterward, the problem persists.

fauxpark, 2019-10-16:
@sizemo what keyboard are you trying to flash?

sizemo, 2019-10-16:
DZ60. I ended up downloading 0.0.12 on a whim and the Terminal command worked, but it's completely nonfunctional on 0.0.13 for me. I put in the command, it doesn't throw back any error or anything, but I start the Toolbox back up and it crashes when you try to flash. 

This is my first go at mechanical keyboards so there's certainly a possibility it was user error, or maybe I'm an idiot and it says something about that in the release notes for 0.0.13 that I just missed. After I downloaded 0.0.12 everything's working fine and I've got my board set up just right now.

fauxpark, 2019-10-16:
Unless you were running version 0.0.10 or earlier, the command is not necessary.
Additionally, it's not relevant for the regular DZ60 - not DZ60RGB - as it's an AVR board and will use dfu-programmer to flash, instead of dfu-util (which is for ARM).

As for the crashing on 0.0.13, are you running macOS Catalina?

sizemo, 2019-10-17:
Yeah, I am. Is there a different problem with Catalina? I couldn't find any other threads that suggested that anything other than this was the problem.

fauxpark, 2019-10-17:
There are two issues with the latest version of the Toolbox on Catalina. The first is that flashing Pro Micros (with avrdude) is completely broken - this is affecting Arduino as well, there's nothing we can do to work around it.
The other problem has to do with dfu-programmer. It works fine, but crashes when trying to print the results. The PR I linked above adds a sanity check to prevent this.