qmk/qmk_toolbox #87: Spaces in paths cause file loading to fail (macOS; others untested)

WillEccles, 2019-06-01:
## Describe the Bug

On macOS Mojave, loading a .hex file with a space in the path results in a path with a %20 in it, which is presumably the error. The file just won't load. However, when the file is moved to a path without a space, there are no problems. If this is macOS specific, I would potentially say that there's a misuse of string/path handling utilities in Obj-C, but since I haven't use it in a while and don't have the time to look into this myself right now, that's the most I can say.

## System Information

 - Operating system: macOS Mojave
 - QMK Toolbox version: 0.0.10

drashna, 2019-06-02:
Windows have the same issue, IIRC. 

So, yeah, we should/need to do some parsing of the path string to fix this.

fauxpark, 2019-06-05:
I can confirm this is a macOS-only problem, which I've fixed in the PR above.

WillEccles, 2019-06-06:
Just had time to look at @fauxpark's changes. Looks like potentially the simplest fix I've ever seen to something, but it's about what I was expecting given the nature of Obj-C. 10/10.

WillEccles, 2019-06-06:
Updated title to reflect the nature of the bug.

PR #90 (2019-06-13): [macOS] Fix loading file:// paths with spaces

## Description

The first thing the Toolbox does when a firmware file is opened is convert its path to an `NSURL`. This has the side effect of percent encoding it. Reasonable, since it *is* a URL, but we don't want it as it just gets passed directly to the flasher tools.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #87
closed 2019-06-13 by drashna