qmk/qmk_toolbox #88: Win 10 reporting both qmk_toolbox.exe and qmk_toolbox_install.exe have malware

alanvgreen, 2019-06-02:
## Describe the Bug

When running version 0.0.10 of qmk_toolkbox_install.exe, Windows warned that it was suspicious.

After installing qmk_toolbox.exe is not present.

When downloading 0.0.10 qmk_toolbox.exe directly, the download fails and the file is not present.

The Windows Threat History reports finding Trojan:Win32/Occamy.C in these files and removing them.

 
## System Information

 - Operating system: Windows 10
 - QMK Toolbox version: 0.0.10

## Additional Context

I was able to download and install 0.0.9 just fine. It has no problems.

RealRadon86, 2019-06-02:
Hi, I have the same problem too...

drashna, 2019-06-02:
/sigh. 

This is because of the BootloadHID.exe file that we use.  It gets flagged as a virus, frequently.  

Try this version: 
https://drive.google.com/open?id=1jQGsPPY3KY8vJXiZNWCStPg1zzwSK-z_

RealRadon86, 2019-06-03:
hi drasha, i am unable to open the link, it says "404. That’s an error. The requested URL was not found on this server. That’s all we know." Is the link public for view/download?

drashna, 2019-06-03:
....... github.   See the underscore at the bottom?  It apparently decided that wasn't part of the URL....  which it is.

fauxpark, 2019-06-14:
I think #89 should have closed this (but it wasn't tagged).

drashna, 2019-07-13:
Sorry, yeah.  And new release coming today.

closed 2019-07-13 by drashna