qmk/qmk_toolbox #93: [Question] Atmega32u4 Connection

Tornquist, 2019-06-18:
I am working on a custom macropad powered by an atmega32u4. I’m struggling to have it recognized by the computer and was wondering if you have any suggestions for verifying the various layers. I realize this isn’t an issue, and am happy to move this if there is a better place to ask.

## Setup

* atmega32u4-au
    * Uses external clock
    * Should come flashed with the atmel bootloader
* Circuit
    * Atmega with datasheet-recommended usb schematic
    * Decoupling capacitors on all power/ground pairs
    * 16 MHz timing crystal
    * Reset on push button. Pulled high normally
    * HWB connected to gnd through 10k r
* Computer
    * macOS with up to date qmk toolbox and dependencies
    * Have used this computer to flash a teensy board and other boards in the past

## Status

* I cannot see the device from dfu-programmer
* No unknown usb devices are listed

## Questions

* Do you have any recommended things to try to verify the device exists/there is a bootloader?
* My understanding is that the out of the box atmel bootloader is fine, and that did-programmer is used through the toolbox to flash the device. Is that understanding correct?

fauxpark, 2019-06-19:
Hi! This is indeed more a place to file issues about the QMK Toolbox itself, but since you're here... 😄

Anyway, I did some sleuthing and found your macropad repo. I don't have EAGLE but I loaded the  schematic into KiCad. It seems you have the D+ and D- lines swapped around on their way to the MCU:
[image]

They should look more like this:
[image]

To answer your questions: On macOS there are no finicky drivers to mess around with so the Toolbox should see it on pressing the reset switch.
dfu-programmer is indeed used by the Toolbox to flash ATmega32U4s, and they come from the factory with the stock Atmel USB DFU bootloader flashed already.

Tornquist, 2019-06-19:
Nice catch, thank you!

I flywired it this morning and it connected as a usb device right away.

closed 2019-06-19 by Tornquist