qmk/qmk_toolbox #97: Can't reset eeprom

kls1991, 2019-06-22:
Windows 10, qmk toolbox 0.0.10
32u4
result like this:
    [ X  ERROR
    Memory write error, use debug for more info.
However,
0.0.9 works......
Sorry for my bad English

fauxpark, 2019-06-22:
Hi, which keyboard is this? Are you still able to flash the firmware with the Toolbox?

kls1991, 2019-06-22:
> Hi, which keyboard is this? Are you still able to flash the firmware with the Toolbox?

Mine is called unigo66.
I can only reset EEPROM with toolbox 0.0.9, I can flash the firmware with 0.0.9 and 0.0.10 both.

fauxpark, 2019-06-23:
Not really sure what's going on here, as it's working perfectly fine for me. You can add the `EEP_RST` keycode to your keymap, which will do the same thing. Please also try to reset the EEPROM from the Toolbox after that.

jibreil, 2019-07-09:
I just came across this issue while helping someone else out with their board, version 0.0.10 kept spitting out the following error when attempting to reset the EEPROM, even after reinstalling toolbox, restarting and replugging the board.

[image]

We were able to successfully reset the EEPROM by using version 0.0.9 as stated by @kls1991

drashna, 2019-07-09:
There was a change in how the EEPROM reset is handled. Unfortunately, it seems to cause issues, in some cases. 

If you download this file, you can flash it:
https://raw.githubusercontent.com/qmk/qmk_toolbox/3d7c9b4c32f1bb7db49e4b0c2a65859fca20bd27/common/atmega32u4_eeprom_reset.hex

But you'll need to flash another HEX file after that.

drashna, 2019-10-01:
We still need to find a good solution for this

internetirl, 2019-10-16:
I got it it to work, but I'm not sure if it was my ordering of events or the fact I used an older version:

https://github.com/qmk/qmk_toolbox/releases/tag/0.0.9

Then:
1.  Opened QMK
2. "Reset" (top right button)
3. Loaded eeprom_reset.hex @drashna linked above
4. "Reset EEPROM" button on bottom right (this worked -- it was not working before)
5. Loaded my custom file

fauxpark, 2019-10-16:
@internetirl there's a couple of things happening here - first of all, the `Reset` button is for *exiting* the bootloader mode (where supported). It doesn't cause the keyboard to jump to the bootloader, and in fact, allowing that is probably not a good idea from a security standpoint.

`eeprom_reset.hex` is actually just a firmware file that does nothing but reset the EEPROM. You flash it like you would any other layout, which means you then have to reflash QMK onto it afterwards.

Prior to version 0.0.10 EEPROM reset works, but it has to do a full chip erase first. This is a limitation of the Atmel factory DFU bootloader, and again requires you to reflash. QMK-DFU (LUFA-DFU) *does* allow you to write to EEPROM without erasing... and it seems I never actually tested #75 on Atmel-DFU.

I think I will have to add back the erase step, as it appears non-trivial to obtain the actual reported device name on Windows to distinguish between Atmel DFU and QMK DFU.

fauxpark, 2021-07-16:
Seems like the above linked PR indicates it fixed this issue, closing.

closed 2021-07-16 by fauxpark