qmk/qmk_toolbox #101: [Feature]

drashna, 2019-07-25:
We should change th eUI so that it denotes that the MCU dropdown is for AVR only. 

## Feature Request Type

- [x] New behavior

fauxpark, 2019-08-13:
I tried appending the text "AVR only" to the label, but it doesn't really fit. I think a decent solution could be only enable the dropdown when a bootloader that requires an MCU parameter to be flashed is detected.

skullydazed, 2019-08-13:
Personally I'd like to see the dropdown go away and detect the MCU based on USB id. We did [this in the old qmk flasher](https://github.com/qmk/qmk_flasher/blob/master/src/MainWindow/app.js#L15) and it seemed to work well.

fauxpark, 2019-08-13:
For DFU that is easy, as each MCU has its own PID. But the Toolbox also supports flashing with AVRISP/USBAsp/USBTiny, which have only a single PID as you're not connecting to the MCU directly.

skullydazed, 2019-08-14:
Would it make sense for the dropdown to be disabled (and maybe blank) unless that PID is detected?

fauxpark, 2019-08-14:
That's what I was getting at. If we find a bootloader that requires avrdude/teensyloader and *isn't* Caterina (I think that is 32U4 only?), the MCU must be manually chosen. Otherwise, we can deduce it from the AVR DFU VID/PID, or ignore it for ARM.

drashna, 2019-10-01:
I think this has been mostly resolved, or at least is handled better now.

PR #164 (2020-01-27): Specify MCU dropdown as AVR only

## Description

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #101
closed 2020-01-27 by drashna