qmk/qmk_toolbox #121: No DFU capable USB device available

eddieedease, 2019-09-15:
QMK toolbox recognizes STM Device in DFU mode, but can't write to it.

`*** STM32 device connected: STM Device in DFU Mode -- 0483:DF11:204B30582032 {36fc9e60-c465-11cf-8056-444553540000}
*** Attempting to flash, please don't remove device
>>> dfu-util.exe -a 0 -d 0483:df11 -s 0x08000000:leave -D "C:\Users\..\Downloads\planck_ez_default_29824f3.bin"
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    
    Cannot open DFU device 0483:df11
    No DFU capable USB device available`

Windows 10, Trying to Flash the Planck. Also tried to install the QMK driver installer but dind't help anything

drashna, 2019-10-01:
This should be handled better now.

First time launching, it should prompt to install the drivers, and the correct drivers for the board.

eddieedease, 2019-10-03:
Can't get it to work unfortunately. There is only a Planck I have which gives that error.  All other boards I have work perfectly. Guess it must be something wrong with this particulae board.

closed 2019-10-03 by eddieedease

fauxpark, 2019-10-03:
What are the other boards you have? It may be that they are all AVR boards and you have the AVR drivers installed, but not the STM32 (ARM) one.

This page should help: https://docs.qmk.fm/#/driver_installation_zadig

eddieedease, 2019-10-04:
I tested it with the NIU Mini 40 & own build Ergodox.. I used Zadig and it says 'installed succesfully', but the qmk toolbox keeps giving error that there is no DFU capable device etc. (But I can see Planck connected and it goes into bootloader mode, it just won't write to it)

I will be trying another PC with W10 soon, will post here if I find anything new.

eddieedease, 2019-11-11:
Just thought about giving it another go. This time it installed the drivers and I now can write to the board. Don't really know what went wrong last time.

TheToteGoat, 2019-11-18:
Any updates on this? I'm having the same issue while trying to flash onto a ProtonC.

fauxpark, 2019-11-18:
@TheToteGoat have you read through the docs page I posted above?

TheToteGoat, 2019-11-19:
I actually found that page shortly after submitting my comment. Zadig worked perfectly! Thank you.

ziqian-L, 2023-11-18:
我想我找到了较为简单的解决方法
首先打开windows的“设备管理器”，然后找到“通用串行总线控制器”，展开，右键“STM DFU”，删除设备驱动，最后使用qmk烧录