qmk/qmk_toolbox #126: Preonic Rev3 default Keymap file extension incorrect

RustyJonez, 2019-10-12:
## Describe the Bug
When downloading a copy of the default keymap for a Rev3 Preonic, the url points to an incorrect file extension.

Currently:

https://qmk.fm/compiled/preonic_rev3_default.hex

Should be:

https://qmk.fm/compiled/preonic_rev3_default.bin

## System Information

 - Operating system: Win 7 64bit
 - QMK Toolbox version: 11

## Additional Context

drashna, 2019-10-14:
I thought I had an issue open for this already, but this is a known issue. 

@fauxpark any ideas?

fauxpark, 2019-10-14:
Well, there is https://github.com/qmk/qmk_toolbox/issues/62.

I don't know if there is a way to download the file with the correct extension short of using the API.

RustyJonez, 2019-10-15:
FWIW, the .bin file can still be downloaded in a web browser and loaded into Toolbox as a workaround for the time being.

fauxpark, 2019-11-11:
I thought of a very simple, but rather ugly, hack:
[image]
[image]

It might be acceptable until the new API can provide this information, there is no ETA on that. I'll see if I can do the same for Windows.

PR #133 (2019-11-29): Precompiled firmware download: try .bin if .hex not found

## Description

Replaces the `.hex` extension with `.bin` if the attempt to download the former failed. Bit of a hack, but it works 👍

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #126
* Closes #62
closed 2019-11-29 by fauxpark