qmk/qmk_toolbox #147: Toolbox Won't launch

archiedevries, 2019-12-16:
Windows says that it is an unsafe file and even after selecting launch anyway the application still does not load. Have tried both the executable and the installer version and whenever I try to launch it absolutly nothing happens. Does anyone have any idea on what I can do to get it working?

drashna, 2019-12-30:
Could you try the newest version, 0.0.15?

JackFennimore, 2020-01-28:
i've tried multiple versions and they all have the same issue on Windows 10. i've tried the same versions on macOS and they work, so it seems like Windows is blocking the launch by default.

fauxpark, 2020-01-29:
It might be Windows Defender, or some other AV software that is blocking it. Double check there, and see if you can add an exception.

FruitJuice, 2020-06-09:
I'm having this exact same issue. On the newest version, and trying to open the Toolbox, just nothing happens at all. I don't see it launching or that stuff, I tried running as admin and in few different compatibility modes. Don't have any anti-virus installed. Windows defender doesn't seem triggered either. I'm using the CU80 Keyboard which is supported by QMK, because I had previously found some forums saying it won't even launch without a supported keyboard connected.

bepstein111, 2020-06-27:
Same issue. Thought it was my Logitech G13, so I unplugged it, still won't open. Tried standalone and installed version, same issue with both.

mattly, 2020-07-01:
I'm seeing the same issue. Windows Defender complained about the app when I first launched it after installing, but it did at least launch. Every attempt to launch since, the app will not load. I've even tried from the `cmd` prompt, and there's no output at all.

mattly, 2020-07-01:
After downloading the executable instead of the installer, and using that, it works fine. No idea what's going on there.

Bpendragon, 2020-07-06:
I've tried throwing most tricks in the book, but Windows 10 Pro (1909) refuses to run either from direct download or installed location. Normal, Admin, `cmd` and PowerShell all fail to get a result beyond dropping some stuff in the event viewer (and I've also tried scanning it with Windows Defender, making sure defender was off temporarily, etc.)

Each attempt to open poops two items in the Event Viewer a `.NET Runtime` error, and an application error, example logs below:

.NET Runtime

```
Application: qmk_toolbox.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Management.ManagementException
   at System.Management.ManagementException.ThrowWithExtendedInfo(System.Management.ManagementStatus)
   at System.Management.ManagementEventWatcher.Start()
   at QMK_Toolbox.MainWindow.StartManagementEventWatcher(System.String)
   at QMK_Toolbox.MainWindow.StartListeningForDeviceEvents()
   at QMK_Toolbox.MainWindow..ctor(System.String)
   at QMK_Toolbox.Program.Main(System.String[])
```

Application Error:

```
Faulting application name: qmk_toolbox.exe, version: 0.0.18.0, time stamp: 0x5eb88c12
Faulting module name: KERNELBASE.dll, version: 10.0.18362.815, time stamp: 0x67005c2a
Exception code: 0xe0434352
Fault offset: 0x00114402
Faulting process id: 0x346e8
Faulting application start time: 0x01d653659b71fae0
Faulting application path: E:\<REDACTED>\qmk_toolbox.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 7e95b412-2af0-4322-80c5-289182f38aa2
Faulting package full name: 
Faulting package-relative application ID: 
```

(note file path is correct in log, just modified here to protect personal data)

FSund, 2020-07-10:
I'm having the same issue, and getting the same errors in Event Viewer as above. 

qmk_toolbox appears briefly in Task Manager, but never opens properly. I'm on a company provided laptop, so I'm suspecting some strict antivirus/antimalware rules, or some Windows policies in place.

Fieel, 2020-07-10:
I am having the same exact problems: qmk both portable and installer won't launch and nothing happens at all. I don't have any Antivirus software and already tried turning off Windows Defender with no luck

sigprof, 2020-07-12:
@Bpendragon 

> Exception Info: System.Management.ManagementException
>    at System.Management.ManagementException.ThrowWithExtendedInfo(System.Management.ManagementStatus)
>    at System.Management.ManagementEventWatcher.Start()
>    at QMK_Toolbox.MainWindow.StartManagementEventWatcher(System.String)

An attempt to start the watcher for device connect/disconnect events fails with some exception, but the event log does not capture enough information about that exception. Probably [StartManagementEventWatcher](https://github.com/qmk/qmk_toolbox/blob/03469f965565a967e886f98bfafa6c872b4aa10e/windows/QMK%20Toolbox/MainWindow.cs#L633-L638) should have some code which catches the exception and shows some information from it (maybe even complete `.ToString()` dump). Not sure whether continuing further in that state would make sense (most toolbox functions won't work without notifications about new USB devices).

sigprof, 2020-07-12:
@Fieel 

> Exception Info: System.ArgumentOutOfRangeException
> 
> Server stack trace: 
>    at System.Array.Resize[T](T[]& array, Int32 newSize)
>    at HidLibrary.HidReport..ctor(Int32 reportSize, HidDeviceData deviceData)
>    at HidLibrary.HidDevice.ReadReport(Int32 timeout)

You have a different issue — the crash in inside HidLibrary. From looking at the [HidReport constructor](https://github.com/mikeobrien/HidLibrary/blob/v3.2.46.0/src/HidLibrary/HidReport.cs#L17-L41), apparently one way to make it crash with `ArgumentOutOfRangeException` is to pass 0 as `reportSize`, and the caller ([`HidDevice.ReadReport(Int32 timeout)`](https://github.com/mikeobrien/HidLibrary/blob/v3.2.46.0/src/HidLibrary/HidDevice.cs#L161-L164)) passes `Capabilities.InputReportByteLength` in that argument; that value comes from the `InputReportByteLength` field in the [`HIDP_CAPS`](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/hidpi/ns-hidpi-_hidp_caps) structure obtained from the Windows HID API. So this crash can be caused by attempting to read an input report from a HID device which does not declare any input reports in its HID descriptor. Such devices actually exist in the wild, and looks like I even have one on my motherboard (Gigabyte X570 AORUS ELITE, the device is probably related to “RGB FUSION 2.0” which I never tried to use):

```
$ sudo lsusb -v -d 048D:8297

Bus 001 Device 003: ID 048d:8297 Integrated Technology Express, Inc. ITE Device(8595)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0x8297 
  bcdDevice            0.03
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8595)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      46
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 21 lines cut ...]
# 0x06, 0x89, 0xff,              // Usage Page (Vendor Usage Page 0xff89) 23
# 0x09, 0xcc,                    // Usage (Vendor Usage 0xcc)           26
# 0xa1, 0x01,                    // Collection (Application)            28
# 0x85, 0xcc,                    //  Report ID (204)                    30
# 0x09, 0x01,                    //  Usage (Vendor Usage 0x01)          32
# 0x15, 0x00,                    //  Logical Minimum (0)                34
# 0x26, 0xff, 0x00,              //  Logical Maximum (255)              36
# 0x75, 0x08,                    //  Report Size (8)                    39
# 0x95, 0x3f,                    //  Report Count (63)                  41
# 0xb1, 0x00,                    //  Feature (Data,Arr,Abs)             43
# 0xc0,                          // End Collection                      45
# 
R: 46 06 89 ff 09 10 a1 01 85 5a 09 01 15 00 26 ff 00 75 08 95 10 b1 00 c0 06 89 ff 09 cc a1 01 85 cc 09 01 15 00 26 ff 00 75 08 95 3f b1 00 c0
N: device 0:0
I: 3 0001 0001
```

(Why that thing declares itself as a boot keyboard is beyond my understanding; as you can see, the report descriptor contains only two feature reports and no input or output reports.)

However, the next question is where an attempt to read the nonexistent input report is performed. The toolbox code apparently tries to read input reports from devices which have the usage page 0xFF31 and code 0x0074 (the debugging console). But those vendor-defined codes are not registered anywhere, so in theory it is possible that some vendor decided to use exactly these codes in their devices, but for a completely different purpose. Apparently the proper way to use the vendor-defined range is to check the USB vendor ID in addition to the vendor-defined usage page and code, but this is not possible for QMK, because it uses lots of different USB vendor ID codes (most of which are not even actually registered properly).

Probably the easiest solution to this problem (if it is really caused by a device with no input reports and the vendor-defined usage page/code collision) is to add one more check to [`GetListableDevices`](https://github.com/qmk/qmk_toolbox/blob/0.0.18/windows/QMK%20Toolbox/MainWindow.cs#L437-L441):
```
                .Where(device => device.Capabilities.InputReportByteLength > 0)
```

sigprof, 2020-07-12:
@Fieel 
I have been able to reproduce a crash with exactly the same backtrace as yours by attaching an ATmega32u4 flashed with a slightly modified “GenericHID” LUFA example code:

```diff
diff --git a/Demos/Device/LowLevel/GenericHID/Descriptors.c b/Demos/Device/LowLevel/GenericHID/Descriptors.c
index f47a406c0..6bc55e2c8 100644
--- a/Demos/Device/LowLevel/GenericHID/Descriptors.c
+++ b/Demos/Device/LowLevel/GenericHID/Descriptors.c
@@ -45,15 +45,15 @@
  */
 const USB_Descriptor_HIDReport_Datatype_t PROGMEM GenericReport[] =
 {
-	HID_RI_USAGE_PAGE(16, 0xFF00), /* Vendor Page 0 */
-	HID_RI_USAGE(8, 0x01), /* Vendor Usage 1 */
+	HID_RI_USAGE_PAGE(16, 0xFF31), /* Vendor Page 0x31 */
+	HID_RI_USAGE(8, 0x74), /* Vendor Usage 0x74 */
 	HID_RI_COLLECTION(8, 0x01), /* Vendor Usage 1 */
 	    HID_RI_USAGE(8, 0x02), /* Vendor Usage 2 */
 	    HID_RI_LOGICAL_MINIMUM(8, 0x00),
 	    HID_RI_LOGICAL_MAXIMUM(8, 0xFF),
 	    HID_RI_REPORT_SIZE(8, 0x08),
 	    HID_RI_REPORT_COUNT(8, GENERIC_REPORT_SIZE),
-	    HID_RI_INPUT(8, HID_IOF_DATA | HID_IOF_VARIABLE | HID_IOF_ABSOLUTE),
+	    HID_RI_FEATURE(8, HID_IOF_DATA | HID_IOF_VARIABLE | HID_IOF_ABSOLUTE),
 	    HID_RI_USAGE(8, 0x03), /* Vendor Usage 3 */
 	    HID_RI_LOGICAL_MINIMUM(8, 0x00),
 	    HID_RI_LOGICAL_MAXIMUM(8, 0xFF),
diff --git a/Demos/Device/LowLevel/GenericHID/GenericHID.c b/Demos/Device/LowLevel/GenericHID/GenericHID.c
index 73856d0e0..0ec1d44f6 100644
--- a/Demos/Device/LowLevel/GenericHID/GenericHID.c
+++ b/Demos/Device/LowLevel/GenericHID/GenericHID.c
@@ -232,6 +232,7 @@ void HID_Task(void)
 		Endpoint_ClearOUT();
 	}
 
+#if 0
 	Endpoint_SelectEndpoint(GENERIC_IN_EPADDR);
 
 	/* Check to see if the host is ready to accept another packet */
@@ -249,5 +250,6 @@ void HID_Task(void)
 		/* Finalize the stream transfer to send the last packet */
 		Endpoint_ClearIN();
 	}
+#endif
 }
 
```
Commands to build and flash this “toolbox killer” code (I used a keyboard with easily accessible Reset as a testbed; different flashing commands will be needed for a Pro Micro):
```sh
make MCU=atmega32u4 BOARD=NONE F_CPU=16000000
dfu-programmer atmega32u4 erase
dfu-programmer atmega32u4 flash GenericHID.hex
dfu-programmer atmega32u4 reset
```

This case is also similar to the issue #65, but in that case the HID descriptor seems to declare an input report, so that issue could be a different one (or the actual code in that XD002 board did not match the provided sources).

Fieel, 2020-07-16:
@sigprof you are absolutely correct, in fact after some physical debugging (unplugging all USB devices) I found the culprit, this little 2-key PCB I bought from drop.com 6 months ago: https://drop.com/buy/xd002-hot-swappable-2-keys-aluminum-macro-pad.

Unplugging the device fixes all problems, QMK toolbox launches as expected and I can flash no problem.

Bpendragon, 2020-07-16:
> @Bpendragon
>
>
> An attempt to start the watcher for device connect/disconnect events fails with some exception, but the event log does not capture enough information about that exception. Probably [...]

Interestingly I had notifications on, and they worked fine. 

I had to install some windows updates involving a machine restart, and that fixed it and it's running fine now. Which is odd because it wasn't working even with a machine restart before.

fauxpark, 2020-10-01:
I believe as of the above PR by sigprof, this is fixed? Can this be closed?

fauxpark, 2021-07-16:
I don't think we've had any more reports of this particular issue, closing.

closed 2021-07-16 by fauxpark