qmk/qmk_toolbox #150: [Feature Request] Flashing handedness with QMK Toolbox

splitkb, 2019-12-23:
The QMK Toolbox is a great tool that makes flashing keyboards more accessible to novice users. One thing that currently cannot be done with the toolbox is to [flash handedness](https://docs.qmk.fm/#/config_options?id=setting-handedness) to split keyboards.

I think this feature could be useful when exposed in the Toolbox, so I decided to open a request to hear your opinions on this.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

drashna, 2019-12-30:
This shouldn't be necessary, anymore.   zvecr  added `MAGIC_EE_HANDS_LEFT` and `MAGIC_EE_HANDS_RIGHT` keycodes, that you can use to set this. 

Though, it would still be useful, yeah.

n8gray, 2020-01-10:
Speaking as a relative newb, `MAGIC_EE_HANDS_*` is nice but I'd much rather have it in the toolbox.  The toolbox is really easy to use, even if you know very little about flashing firmware. The idea of setting up bootmagic and binding keys to keycodes just so I can set the handedness of my halves seems kind of painful TBH.

fauxpark, 2021-07-22:
@n8gray @splitkb see the above PR :)

PR #287 (2021-07-23): Add support for setting handedness in EEPROM

## Description

The Toolbox can now set `EE_HANDS`!

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #150
closed 2021-07-23 by fauxpark