qmk/qmk_toolbox #152: Crash on MacOS Mojave while trying to flash

clbx, 2019-12-30:
## Describe the Bug

QMK Toolbox Crashes when trying to flash Massdrop Alt board

## System Information

 - Operating system: MacOS 10.14.6 Mojave
 - QMK Toolbox version: 0.0.15 (1)

## Additional Context

On flashing (Both Massdrop Default and my own FW) QMK Immediately Crashes. The Problem holds [this](https://pastebin.com/s89JQzUp). However this seems like the most important part

```
Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'launch path not accessible'
abort() called
terminating with uncaught exception of type NSException
```

## What I've tried so far

I've given ``avrdude dfu-programer dfu-util teensy_loader_cli`` execute permissions, I'm not sure which one QMK uses for the Alt, but this didn't seen to fix the issue.

fauxpark, 2019-12-30:
Sorry about that! Yeah, we had a similar problem with dfu-util not having the executable bit set in Git, which propagated out to the final build artifact, so the Toolbox couldn't run it. Seems like this is the same issue but with the Massdrop Loader tool. The PR linked above will fix it, but it may be a short while before there's a new release. For now, you can just `chmod +x mdloader_mac` in the QMK Toolbox.app.

closed 2019-12-30 by drashna

clbx, 2019-12-30:
Giving execute permissions to the mdloader did not fix the issue.

clbx, 2020-01-01:
@fauxpark @drashna setting the executable bit still crashes the program.

fauxpark, 2020-01-02:
Ah, I see what the problem is now. The link to it in the Xcode project was set up incorrectly. This time you'll have to wait for a new release, sorry :/

PR #153 (2019-12-30): Set mdloader_mac executable bit

## Description

Not sure how this slipped by me again... just reinforces the need for something like #86.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #152
PR #157 (2020-01-03): Fix path to mdloader in Xcode project file

## Description

As well as not being executable the reference to it in the pbxproj was different to the references for avrdude, dfu-programmer etc.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #152 (properly this time)