qmk/qmk_toolbox #160: avrdude.exe: option requires an argument -- P

nalanj, 2020-01-21:
## Describe the Bug

When attempting to flash an atmega32u4 on Windows I'm getting the error "avrdude.exe: option requires an argument -- P". I've seen this in previous versions as well, and thought it was fixed per #60, but flashing from 0.0.15 still yields the error.

## System Information

 - Operating system: Windows 10
 - QMK Toolbox version: 0.0.15

## Additional Context

```
*** QMK Toolbox (https://qmk.fm/toolbox)
    Supported bootloaders:
     - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
     - ARM DFU (STM32, Kiibohd) via dfu-util (http://dfu-util.sourceforge.net/)
     - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
     - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
    Supported ISP flashers:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
     - USBasp (AVR ISP)
*** Caterina device connected: (Standard system devices) USB Input Device (2341:8037:0100)
*** Caterina device connected: Arduino LLC (www.arduino.cc) Arduino Micro (COM3) (2341:8037:0100)
*** Caterina device connected: (Standard USB Host Controller) USB Composite Device (2341:8037:0100)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"C:\Users\alanj\Downloads\lets_split_rev2_default_e0880824.hex":i -P 
    C:\Users\alanj\AppData\Local\QMK\QMK Toolbox\0.0.15\avrdude.exe: option requires an argument -- P
    Usage: avrdude.exe [options]
    Options:
      -p <partno>                Required. Specify AVR device.
      -b <baudrate>              Override RS-232 baud rate.
      -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
      -C <config-file>           Specify location of configuration file.
      -c <programmer>            Specify programmer type.
      -D                         Disable auto erase for flash memory
      -i <delay>                 ISP Clock Delay [in microseconds]
      -P <port>                  Specify connection port.
      -F                         Override invalid signature check.
      -e                         Perform a chip erase.
      -O                         Perform RC oscillator calibration (see AVR053). 
      -U <memtype>:r|w|v:<filename>[:format]
                                 Memory operation specification.
                                 Multiple -U options are allowed, each request
                                 is performed in the order specified.
      -n                         Do not write anything to the device.
      -V                         Do not verify.
      -u                         Disable safemode, default when running from a script.
      -s                         Silent safemode operation, will not ask you if
                                 fuses should be changed back.
      -t                         Enter terminal mode.
      -E <exitspec>[,<exitspec>] List programmer exit specifications.
      -x <extended_param>        Pass <extended_param> to programmer.
      -y                         Count # erase cycles in EEPROM.
      -Y <number>                Initialize erase cycle # in EEPROM.
      -v                         Verbose output. -v -v for more.
      -q                         Quell progress output. -q -q for less.
      -l logfile                 Use logfile rather than stderr for diagnostics.
      -?                         Display this usage.
    
    avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>
```

fauxpark, 2020-01-21:
It looks like you've got the Arduino drivers installed - these really aren't needed for Caterina, so I think uninstalling the devices in Device Manager will help. The Toolbox should only be detecting a single device, not three.

nalanj, 2020-01-21:
That did it - thanks for the help!

closed 2020-01-21 by nalanj