qmk/qmk_toolbox #162: Crashes when flash Massdrop alt keyboard on macOS Catalina

zizhenc, 2020-01-22:
I want to use QMK toolbox directly flash my Drop (formerly Massdrop) Alt High profile keyboard on macOS Catalina.
I tried both compiled files from Massdrop configurator website and QMK Configurator website. The crash problem always happen when you try to click the flash button, then the flash cannot be done through QMK toolbox on macOS Catalina.

I can use the mdloader to flash the Alt keyboard without any issues.

fauxpark, 2020-01-22:
Could you please provide more info? What's the exact error you're getting?
A screenshot or crash log would help a lot.

zizhenc, 2020-01-22:
> Could you please provide more info? What's the exact error you're getting?
> A screenshot or crash log would help a lot.
The crash log of the QMK toolbox.app is here:

Process:               QMK Toolbox [790]
Path:                  /Applications/QMK Toolbox.app/Contents/MacOS/QMK Toolbox
Identifier:            fm.qmk.toolbox
Version:               0.0.15 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QMK Toolbox [790]
User ID:               501

Date/Time:             2020-01-22 04:33:30.727 -0500
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Bridge OS Version:     4.2 (17P2551)
Anonymous UUID:        AD7FAB9E-2812-13CC-1806-0A29BE91679D

Time Awake Since Boot: 1200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'launch path not accessible'
abort() called
terminating with uncaught exception of type NSException

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff319548ab __exceptionPreprocess + 250
1   libobjc.A.dylib                     0x00007fff67bc5805 objc_exception_throw + 48
2   Foundation                          0x00007fff34020e71 -[NSConcreteTask launchWithDictionary:error:] + 5234
3   QMK Toolbox                         0x0000000103f0934a QMK Toolbox + 17226
4   QMK Toolbox                         0x0000000103f0a9be QMK Toolbox + 22974
5   QMK Toolbox                         0x0000000103f095ae QMK Toolbox + 17838
6   QMK Toolbox                         0x0000000103f0b349 QMK Toolbox + 25417
7   libdispatch.dylib                   0x00007fff68eda50e _dispatch_client_callout + 8
8   libdispatch.dylib                   0x00007fff68edc6c0 _dispatch_continuation_pop + 414
9   libdispatch.dylib                   0x00007fff68eec3c4 _dispatch_source_invoke + 2084
10  libdispatch.dylib                   0x00007fff68ee5a82 _dispatch_main_queue_callback_4CF + 783
11  CoreFoundation                      0x00007fff31917e00 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
12  CoreFoundation                      0x00007fff318d7b8a __CFRunLoopRun + 2370
13  CoreFoundation                      0x00007fff318d6bd3 CFRunLoopRunSpecific + 499
14  HIToolbox                           0x00007fff3042d65d RunCurrentEventLoopInMode + 292
15  HIToolbox                           0x00007fff3042d39d ReceiveNextEventCommon + 600
16  HIToolbox                           0x00007fff3042d127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
17  AppKit                              0x00007fff2ea9eeb4 _DPSNextEvent + 990
18  AppKit                              0x00007fff2ea9d690 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
19  AppKit                              0x00007fff2ea8f3ae -[NSApplication run] + 658
20  AppKit                              0x00007fff2ea61775 NSApplicationMain + 777
21  libdyld.dylib                       0x00007fff68f337fd start + 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6907a7fa __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff69137bc1 pthread_kill + 432
2   libsystem_c.dylib             	0x00007fff69001a1c abort + 120
3   libc++abi.dylib               	0x00007fff6609fbe8 abort_message + 231
4   libc++abi.dylib               	0x00007fff6609fd9c demangling_terminate_handler() + 262
5   libobjc.A.dylib               	0x00007fff67bc778a _objc_terminate() + 96
6   libc++abi.dylib               	0x00007fff660acdc7 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff660acd79 std::terminate() + 41
8   libdispatch.dylib             	0x00007fff68eda522 _dispatch_client_callout + 28
9   libdispatch.dylib             	0x00007fff68edc6c0 _dispatch_continuation_pop + 414
10  libdispatch.dylib             	0x00007fff68eec3c4 _dispatch_source_invoke + 2084
[... 377 lines cut ...]
shared memory                      656K       19 
===========                     =======  ======= 
TOTAL                              1.3G      785 
TOTAL, minus reserved VM space   972.6M      785 

Model: MacBookPro15,2, BootROM 1037.60.58.0.0 (iBridge: 17.16.12551.0.0,0), 4 processors, Quad-Core Intel Core i7, 2.8 GHz, 16 GB, SMC 
Graphics: kHW_IntelIrisGraphics655Item, Intel Iris Plus Graphics 655, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR3, 2133 MHz, SK Hynix, -
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR3, 2133 MHz, SK Hynix, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Oct 26 2019 10:18:37 version 9.112.2.0.32.5.40 FWID 01-66dd2dcb
Bluetooth: Version 7.0.2f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: Hub
USB Device: Communication Device
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2

fauxpark, 2020-01-22:
Ah, this is the same issue as in #152 - the problem's been fixed but needs a new release to be published. It's been a good while so I think that should happen fairly soon. Sorry about that!

zizhenc, 2020-01-22:
Thanks for the quick reply, hope to receive the new release soon!

closed 2020-01-22 by zizhenc

drashna, 2020-01-22:
Released as 0.0.16, actually.