qmk/qmk_toolbox #170: No key input after flashing device (lfk78)

CosmosAtlas, 2020-02-21:
I'm having trouble flashing a lfk78 pcb (rev unknown)

The problem I'm facing now is the keyboard won't register any keys when plugged in. The computer will detect its a keyboard (in Mac & Linux)

The keyboard's controller on the pcb is 90USB1286.

What I have tried so far:

On Linux: 
- Directly compiling a firmware without any edits the make lfkeyboards/lfk78:default:dfu will stuck at 5s loop. (Some lookup seems to suggest different controller used by revisions) 
- Setting the rules.mk in the root keyboards/lfkeyboards/lfk78 folder to use rev. D (which is what windows told me is when I first plugged in). 
- Randomly flashing all other versions (C,D,E,...,H).

Windows: 
- Trying the same thing with windows by QMK toolbox. Which won't flash, will give device not found. Using hex compiled from linux and setting to 90USB1286.

Mac: 
- Setting the rules.mk in the root keyboards/lfkeyboards/lfk78 folder to use rev. D (which is what windows told me is when I first plugged in). (Same as second step with Linux)

Now I'm stuck and don't know what to try.

noroadsleft, 2020-02-22:
@fauxpark Any ideas?

fauxpark, 2020-07-06:
@CosmosAtlas I refactored the LFK78 a while back to not require editing rules.mk to specify the revision, it's been split into `revb`, `revc` (includes rev D) and `revj`. `revc` is the only one which uses an AT90USB1286.

What do you see in Device Manager or lsusb when you try to reset the board?

CosmosAtlas, 2020-07-06:
@fauxpark IIRC based on the version, settings.. etc, it sometimes show `lfk78` and that's it. 

Anyhow, when I send the keyboard to someone for service (~early March) after having a chat with MechMerlin on discord reaching a point where a multimeter is required for debugging which I have no knowledge on. 

Some attempts have been made trying to fix it but no luck till now. I guess I might just end up salvaging it and eating up the cost.

fauxpark, 2021-07-16:
@CosmosAtlas any update on this, or can it be closed?

CosmosAtlas, 2021-07-16:
@fauxpark I believe the device is still not working, but I don't have it on my hands ATM. I'll go ahead and close the issue as I don't think it'll be solved anytime in the near future as well.

CosmosAtlas, 2021-12-03:
@fauxpark I'm not sure if you are still around. But I got my keyboard bac know (after a long time but yes)

The problem still persists with the newest version of QMK.

lsusb shows the following

```
Bus 001 Device 013: ID feed:6060 LFKeyboards LFK78
```

In DFU mode (i.e., clicking the reset button)
```
Bus 001 Device 015: ID 03eb:2ffb Atmel Corp. at90usb AVR DFU bootloader
```

The chip reads `90USB1286AU`, so I must be flashing with the right version.

Any help is appreciated. : -)

CosmosAtlas, 2022-03-11:
Closed since I no longer have the PCB anymore.

closed 2022-03-11 by CosmosAtlas