qmk/qmk_toolbox #187: Disable MCU selection box if BIN file is loaded

nooges, 2020-04-16:
## Feature Request Type

- [ ] Core functionality
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

For people flashing to non-AVR devices, the MCU dropdown can still be confusing, even though it says `(AVR only)`. Might be a good idea to disable the box if a .bin file is loaded in or do some other detection method if someone has a better idea.

fauxpark, 2020-04-16:
I think we will need to refactor the bootloader detection process so that it only picks up the first device it finds. Then depending on which bootloader that is, it will be much simpler to set the state of the dropdown.

nooges, 2020-04-17:
Yeah, I'd prefer that, my other thought along the same lines had been to check the `canFlash` status.

drashna, 2020-04-17:
What about the tada, that uses a bin?

yanfali, 2020-04-17:
> What about the tada, that uses a bin?

tada doesn't use toolbox, it uses file manager

skullydazed, 2020-05-15:
I feel like it should be disabled until it's needed. It's pretty confusing and only actually needed in a limited number of circumstances.