qmk/qmk_toolbox #188: avrdude.exe: option requires an argument -- P

truthbeliever, 2020-04-24:
## Describe the Bug

When attempting to flash an atmega32u4 on Windows I'm getting the error avrdude.exe: option requires an argument -- P. I've seen this in previous versions as well, and thought it was fixed per #160,
It seems to be only one device as opposed to #160.
It seems it needs some port number...
Any hint?
Txs

## System Information

 - Operating system: Win 8.1
 - QMK Toolbox version:  0.0.17

## Additional Context

*** QMK Toolbox 0.0.17 (https://qmk.fm/toolbox)
    Supported bootloaders:
     - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
     - ARM DFU (STM32, Kiibohd) via dfu-util (http://dfu-util.sourceforge.net/)
     - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
     - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
    Supported ISP flashers:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
     - USBasp (AVR ISP)
*** Caterina device connected:  Arduino Leonardo (2341:0036:0001)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"C:\Users\schlimel\Desktop\redox_rev1_david.hex":i -P 
    C:\Users\schlimel\AppData\Local\QMK\QMK Toolbox\0.0.17\avrdude.exe: option requires an argument -- P

truthbeliever, 2020-04-25:
installed official Arduino software from
https://www.arduino.cc/
now it works...

closed 2020-04-25 by truthbeliever

fauxpark, 2020-04-25:
You don't need Arduino installed for it to work. The driver Caterina uses is built in to Windows (`usbser`). My guess is there was no driver installed for the device, or it was incorrect, and installing the Arduino IDE "fixed" it by installing their own driver.