qmk/qmk_toolbox #189: Error: The target memory is not blank

gilbertfrancois, 2020-04-28:
I try to flash a hotdox keyboard, but it gives an error and the keyboard has still its old keymap after flashing.

## System Information

 - Operating system: macOS 10.15.4
 - QMK Toolbox version: 0.17

## Log output

```
*** QMK Toolbox 0.0.17 (http://qmk.fm/toolbox)
    Supported bootloaders:
     - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
     - ARM DFU (STM32, Kiibohd) via dfu-util (http://dfu-util.sourceforge.net/)
     - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
     - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
    Supported ISP flashers:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
     - USBasp (AVR ISP)
*** Z.P.Z.O. - HotDox connected -- FEED:E11D
*** Z.P.Z.O. - HotDox disconnected -- FEED:E11D
*** DFU device connected
*** Attempting to flash, please don't remove device
>>> dfu-programmer atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
>>> dfu-programmer atmega32u4 flash /Users/gilbert/Dropbox/Persoonlijk/Library/hotdox/gilbert_v2.hex
    Checking memory from 0x0 to 0x58FF...  Not blank at 0x1.
    The target memory for the program is not blank.
    Use --force flag to override this error check.
>>> dfu-programmer atmega32u4 reset
*** DFU device disconnected
*** Z.P.Z.O. - HotDox connected -- FEED:E11D
  > mcp23017 attached!!!
  > bootmagic scan: ... done.
    Keyboard start.
```

fauxpark, 2020-04-28:
This is a known issue that seems to happen randomly, and we're not sure what causes it. However, ISP flashing the bootloader again appears to fix it.
https://docs.qmk.fm/#/isp_flashing_guide

gilbertfrancois, 2020-04-29:
Thanks @fauxpark. I will give it a try.

evandam, 2020-05-05:
I'm seeing the same issue on one side of my Keebio Iris rev 4. Trying to ISP flash but can't seem to get a successful flash:

```
make keebio/iris/rev4:default:production
. . .

# Flashing keebio_iris_rev4_default_bootloader.hex through QMK toolbox
>>> dfu-programmer atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
>>> dfu-programmer atmega32u4 flash /Users/evan/qmk_firmware/keebio_iris_rev4_default_bootloader.hex
    Bootloader and code overlap.
    Use --suppress-bootloader-mem to ignore
  ! File is too large for device

$ dfu-programmer atmega32u4 flash /Users/evan/qmk_firmware/keebio_iris_rev4_default_bootloader.hex  --suppress-bootloader-mem --debug 1
Hex file error, use debug for more info.
```

Any pointers on how to get around this?

fauxpark, 2020-05-05:
dfu-programmer is not able to ISP flash AVRs. You need to use an actual ISP programmer; you can flash a spare Pro Micro or Teensy with the AVRISP firmware and wire it up to the Iris's ISP header. See the link I posted above.

gilbertfrancois, 2020-05-22:
I have a Sparkfun Bus Pirate v3.6a and a HotDox keyboard with atmega32u4, so I should be ready do the ISP flashing. The ISP flashing guide explains the wiring of a SparkFun Pocket AVR, Teensy 2.0 and Pro Micro, but is missing the wiring of a SparkFun Bus Pirate to a keyboard. Does anyone know how to connect it? I guess it is very similar to the Pocket AVR, except that I don't see a RST pin on the BP.

fauxpark, 2020-07-06:
@gilbertfrancois I'm not sure the Bus Pirate is actually able to do ISP flashing, which would explain why the pinout doesn't line up.

PR #215 (2020-08-20): Use the force when flashing with dfu-programmer

## Description

Same as qmk/qmk_firmware#10070.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #189
closed 2020-08-20 by fauxpark