qmk/qmk_toolbox #190: Is there any way for this app to be seen as safe by Windows/Mac?

amowry, 2020-04-29:
Many thanks for all your work on this! I'm recommending QMK Toolbox as the preferred method for flashing MIDI devices that I make. Some of my customers are concerned that the app is seen as dangerous by Windows Defender, Norton, and Mac OS. I was just curious if there are any ways in which it would be seen as safe? Thanks!

skullydazed, 2020-04-29:
From a high level we need to sign the app to avoid that prompt. That's a chunk of work for both mac and windows that no one has taken on yet. In the short term I don't have a good answer on Windows, but on macOS you can Right Click->Open the first time to bypass the prompt without turning off Gatekeeper. 

We'll talk more internally about whether or not signing is something we have the resources to do.

amowry, 2020-04-29:
Okay, thanks very much for looking into it, and for the fast reply!