qmk/qmk_toolbox #191: Provide instructions to uninstall drivers

GauthierPLM, 2020-04-29:
I could not find any information about uninstalling or removing the drivers from my system when I am done with the Toolbox. Could you provide instructions in the documentation or in the software?
If drivers are uninstalled with the Toolbox, could you mention it in the uninstaller to avoid confusion?

fauxpark, 2020-05-10:
There is no way to uninstall the drivers using qmk_driver_installer (well, libwdi). You just have to uninstall them through Device Manager.

See this issue: https://github.com/pbatard/libwdi/issues/166

fauxpark, 2020-09-15:
As it turns out, there kind of is a way to completely remove the drivers, but it's a little involved and potentially dangerous.

First you must uninstall all related devices in Device Manager (even ones that are not connected). Then you can run `pnputil /enum-drivers` from an elevated Command Prompt, which will list all the `oem#.inf` files installed. Running `pnputil /delete-driver <oem#.inf>` for each driver you want to delete should get rid of them - in this case they will all mention libwdi, I think, and as mentioned, all devices using them must be uninstalled or it will complain.

See the pnputil command syntax: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-command-syntax

Eljefemasao, 2020-10-20:
How to uninstall the driver on Mac ? thanks.

fauxpark, 2020-10-20:
@Eljefemasao there are no drivers on macOS.

Eljefemasao, 2020-10-20:
Thank you for your answer.

I implemented some command following.

And got an error following.

Do you know how to fix it?

Eljefemasao, 2020-10-20:
Thank you for your answer.
I implemented the following.

>make dz60:stephengrier (passed)
>make dz60:stephengrier:dfu (faild)

And got an error following.
dfu-programmer: no device present.

I'd appreciate it If you know the solution and tell me something.

fauxpark, 2020-10-20:
Did you press the reset button on the back of the PCB? System Information.app should show a device named something like "ATm32U4DFU" in the USB section.

Eljefemasao, 2020-10-20:
It worked! 
I've been tried (Function Key + \ ) as a reset button. 
Thank you for your kind Advice!

jart, 2021-06-15:
The drivers compromise system stability so if they can't be uninstalled that's a problem. qmk_driver_installer.exe (#29) doesn't have any source code and it uses many tactics commonly associated with malware, such as embedding other executables inside itself using concealment methods that which be detected by most tools, in addition to going out of its way to detect the presence of a debugger.

tzarc, 2021-06-15:
https://docs.qmk.fm/#/driver_installation_zadig?id=recovering-from-installation-to-wrong-device

fauxpark, 2021-06-15:
Also, https://github.com/qmk/qmk_driver_installer

closed 2021-07-16 by fauxpark