qmk/qmk_toolbox #194: XD75 Mac QMK toolbox dfu-programmer: no device present.

rendomnet, 2020-05-10:
Trying to flash using QMK.

I have put my keyboard to reset mode.

It shows DFU device connected but then when I click flash. it gives me "dfu-programmer: no device present."
[image]

fauxpark, 2020-05-10:
Have you tried updating the Toolbox?

rendomnet, 2020-05-10:
@fauxpark version 0.0.17 (1). Downloaded from here https://github.com/qmk/qmk_toolbox/releases

fauxpark, 2020-05-10:
Do you see anything in System Information, under USB? Also, what version of macOS are you running?

rendomnet, 2020-05-10:
@fauxpark  Mojave 10.14.6
[image]

fauxpark, 2020-05-10:
Okay, what about when it's running as a keyboard?

rendomnet, 2020-05-10:
@fauxpark same. 
But under Windows10 I see new device. But also cant flash.

fauxpark, 2020-05-10:
This isn't a hackintosh build, is it?

rendomnet, 2020-05-10:
@fauxpark it is.

rendomnet, 2020-05-10:
Under Windows same thing. 
[image]

fauxpark, 2020-05-10:
In that case, I guess there is some problem with the USB kext or something, not the Toolbox - I'm not really familiar with Hackintosh stuff, but the complete lack of USB devices at all made me suspicious. With absolutely nothing connected to my MBP I still see two internal hub devices:
[image]

As for the Windows side, you'll probably need to install the drivers. On first run the Toolbox should prompt you to install them, and if that doesn't work run [Zadig](https://zadig.akeo.ie) and make sure the ATm32U4DFU device (the bootloader of your XD75) has the libusb0/libusb-win32 driver (not WinUSB or libusbK)

rendomnet, 2020-05-10:
@fauxpark yeah I have tried Zadig and for some reason when I'm pressing install after several seconds it just closes. :-|

fauxpark, 2020-05-10:
Tried running as admin? Anything in Event Viewer?

rendomnet, 2020-05-10:
@fauxpark yeah running as administrator 
[image]

fauxpark, 2020-05-10:
`tumint.dll` seems to be part of some software called GammaTech TumarCSP, I'm not sure what that is. Looks Russian and shady.

rendomnet, 2020-05-10:
@fauxpark hurray! Uninstalling this software helped!

fauxpark, 2020-09-15:
@rendomnet can this be closed?

closed 2020-10-01 by fauxpark