qmk/qmk_toolbox #198: MCU dropdown should not affect dfu-programmer behavior

skullydazed, 2020-05-15:
When we're using dfu-programmer we know what kind of MCU is attached. The MCU dropdown should be ignored when building a dfu-programmer command, and we should always use the vid/pid detection to determine mcu.

## Feature Request Type

- [ ] Core functionality
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

nooges, 2020-05-16:
Some VID/PID pairs listed here: http://ww1.microchip.com/downloads/en/devicedoc/doc7618.pdf

skullydazed, 2020-05-16:
We had this functionality in the old qmk firmware flasher and the list of pid's that used is here: https://github.com/qmk/qmk_flasher/blob/master/src/MainWindow/app.js#L15