qmk/qmk_toolbox #199: Error while writing

amowry, 2020-05-23:
I've heard from someone using the most recent QMK Toolbox release for Mac who had an error while writing to a 32u4 that appears to have bricked the device. I was just wondering if you have seen this before? It appears to have been writing and then encountered an error. Thanks!
[image]

fauxpark, 2020-05-23:
Make sure you're on the latest version of macOS (there was a previous version of Catalina that avrdude had issues with when talking to the Pro Micro bootloaders; see https://github.com/qmk/qmk_firmware/issues/6133)
Also double check your connections and soldering, just in case the flakiness is coming from there.

If all else fails, you may have to ISP flash the chip: https://docs.qmk.fm/#/isp_flashing_guide

amowry, 2020-05-24:
Okay, thanks for the suggestions!

fauxpark, 2020-07-06:
@amowry any update on this? Did you get it working?

amowry, 2020-07-06:
Thanks for checking in. I believe my customer ended up getting the installation to work on Windows instead, so we never figured out the Mac issue, but it's certainly possible it was user error. I've had a few hundred customers using QMK to update firmware on my device and it seems to work fine for most. Thanks very much for the support!

closed 2020-07-06 by fauxpark