qmk/qmk_toolbox #201: Unable to open app on Mac OS Big Sur

timtraversy, 2020-06-30:
##  Describe the Bug

I'm running the beta of Mac OS Big Sur. I installed the latest release of `QMK.Toolbox.app.zip`,  currently `0.0.18`, and I'm unable to run it. There's a bunch of apps that are broken on Big Sur right now, so I assume some arch change was made that's causing this issue.

https://forums.macrumors.com/threads/big-sur-working-not-working-apps.2242312/

## System Information

 - Operating system: Mac OS Big Sur
 - QMK Toolbox version: 0.0.18

skullydazed, 2020-06-30:
Can you provide detail on what happens when you try to run it? Is there a crash report you can attach?

timtraversy, 2020-06-30:
If I open it from Dock it fails quietly, if I open it from Spotlight I get a `Impossible to open application "QMK Toolbox"`

timtraversy, 2020-06-30:
The only error it's reporting is `No architectures specified in launch archictectures [ NULL ]  for app=file:///Applications/QMK%20Toolbox.app/ which likely is an error.`, so that could be it

timtraversy, 2020-06-30:
But actually, I'm seeing that in the logs for other apps that are launching fine.

timtraversy, 2020-06-30:
Dropbox is failing also, and their log included `ImportError: Framework CoreServices could not be found`

https://www.dropboxforum.com/t5/Dropbox-files-folders/Dropbox-app-doesn-t-start-on-MacOS-Big-Sur-11-0-beta/td-p/431820

skullydazed, 2020-06-30:
Interesting, thanks for the information. One last test, if you find `QMK Toolbox.app` in the Finder, right click it, and select "Open", does that change the behavior?

timtraversy, 2020-06-30:
No, still fails to open that way.

Another thing, from system.log:

```
```

skullydazed, 2020-06-30:
Does it fail silently like the dock, or with a message like spotlight?

timtraversy, 2020-06-30:
It fails silently.

skullydazed, 2020-06-30:
Thanks! I don't know when someone will have time to install the beta and look at this further, but for now we'll leave this open. Anyone who is interested in the state of QMK Toolbox on Big Sur can subscribe to this issue for updates.

nooges, 2020-06-30:
I’ll have the beta installed soon enough to do some investigation.

ehagerty, 2020-07-07:
run it as sudo /Applications/QMK\ Toolbox.app/Contents/MacOS/QMK\ Toolbox

i've seen a lot of apps need this in the past year as a beta tester

timtraversy, 2020-07-07:
That did it for me, thanks @ehagerty! I think we should keep this open until it can be resolved without the command line.

fauxpark, 2020-11-17:
I'm running Big Sur now (11.0.1), and it seems to work just fine as is. I tested flashing a couple of devices and that appears to work too. @timtraversy @ehagerty can you confirm?

ehagerty, 2020-11-17:
@fauxpark thank you very much!

MacBook Pro (15-inch, 2019)
macOS 11.0.1 (20B29)
QMK Toolbox 0.0.20 (1)

Runs as expected, without needing sudo as above.

fauxpark, 2021-07-16:
Closing, Toolbox appears to be running fine on Big Sur so far.

closed 2021-07-16 by fauxpark