qmk/qmk_toolbox #203: option requires an argument -- P

amowry, 2020-07-14:
One of my customers is getting this error on Windows 7. The device is a 32u4 using an Adafruit bootloader, so I had him install the Adafruit drivers, which I know work with avrdude and command line. Should he have let QMK install drivers instead? It's my understanding that Windows 7 needs some kind of driver installed, correct?

Thanks!

*** Auto-flash enabled
*** Caterina device connected:  Adafruit ItsyBit (239A:000D:0001)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p at90usb647 -c avr109 -U flash:w:"C:\Users\DB MacInnes\Desktop\WARBL_1.8.ino.hex":i -P 
    C:\Users\DB MacInnes\AppData\Local\QMK\QMK Toolbox\0.0.17\avrdude.exe: option requires an argument -- P

fauxpark, 2020-07-14:
Caterina bootloaders show up as a standard serial port, so there is no driver needed except for the builtin Windows usbser driver. Make sure the bootloader is using that one in Device Manager.

One other thing I can see is that the MCU dropdown is set wrong - the avrdude command is looking for an AT90USB647, that should be `atmega32u4`.

amowry, 2020-07-14:
Oh, I hadn't noticed the MCU, thanks! I'll have him fix that and try again.

I was under the impression that Windows 7 required a serial port driver for Caterina even though Windows 10 doesn't. That will be ideal if it's not needed in Windows 7 either. 

Thanks again! I'll report back.

fauxpark, 2020-07-14:
You might be right there, actually. It seems like Windows 7 did *not* come with the usbser driver: https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-driver-installation-based-on-compatible-ids

In that case, you could try installing the CDC serial driver onto the bootloader device with [Zadig](https://zadig.akeo.ie).

amowry, 2020-07-14:
It turned out that just selecting the correct MCU fixed the issue. Thanks again for noticing that!

closed 2020-07-14 by amowry