qmk/qmk_toolbox #204: avrdude.exe: option requires an argument -- P

brez1623, 2020-07-15:
When trying to flash a milk pcb I get the error avrdude.exe option requires an argument

 - Operating system: Windows 10 
 - QMK Toolbox version: 0.0.18

*** QMK Toolbox 0.0.18 (https://qmk.fm/toolbox)
    Supported bootloaders:
     - Atmel/LUFA/QMK DFU via dfu-programmer (http://dfu-programmer.github.io/)
     - Caterina (Arduino, Pro Micro) via avrdude (http://nongnu.org/avrdude/)
     - Halfkay (Teensy, Ergodox EZ) via Teensy Loader (https://pjrc.com/teensy/loader_cli.html)
     - ARM DFU (STM32, Kiibohd) via dfu-util (http://dfu-util.sourceforge.net/)
     - Atmel SAM-BA (Massdrop) via Massdrop Loader (https://github.com/massdrop/mdloader)
     - BootloadHID (Atmel, PS2AVRGB) via bootloadHID (https://www.obdev.at/products/vusb/bootloadhid.html)
    Supported ISP flashers:
     - USBTiny (AVR Pocket)
     - AVRISP (Arduino ISP)
     - USBasp (AVR ISP)
*** Caterina device connected: Microsoft USB Serial Device (COM4) (2341:8037:0100)
*** Caterina device connected: (Standard USB Host Controller) USB Composite Device (2341:8037:0100)
*** Caterina device connected: (Standard system devices) USB Input Device (2341:8037:0100)
*** HID console connected: CannonKeys Instant60 (CA04:1600:0001)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"E:\Downloads\spaceman_2_milk_layout_mine.hex":i -P 
    C:\Users\Tbrezniak\AppData\Local\QMK\QMK Toolbox\0.0.18\avrdude.exe: option requires an argument -- P
    Usage: avrdude.exe [options]
    Options:
      -p <partno>                Required. Specify AVR device.
      -b <baudrate>              Override RS-232 baud rate.
      -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
      -C <config-file>           Specify location of configuration file.
      -c <programmer>            Specify programmer type.
      -D                         Disable auto erase for flash memory
      -i <delay>                 ISP Clock Delay [in microseconds]
      -P <port>                  Specify connection port.
      -F                         Override invalid signature check.
      -e                         Perform a chip erase.
      -O                         Perform RC oscillator calibration (see AVR053). 
      -U <memtype>:r|w|v:<filename>[:format]
                                 Memory operation specification.
                                 Multiple -U options are allowed, each request
                                 is performed in the order specified.
      -n                         Do not write anything to the device.
      -V                         Do not verify.
      -u                         Disable safemode, default when running from a script.
      -s                         Silent safemode operation, will not ask you if
                                 fuses should be changed back.
      -t                         Enter terminal mode.
      -E <exitspec>[,<exitspec>] List programmer exit specifications.
      -x <extended_param>        Pass <extended_param> to programmer.
      -y                         Count # erase cycles in EEPROM.
      -Y <number>                Initialize erase cycle # in EEPROM.
      -v                         Verbose output. -v -v for more.
      -q                         Quell progress output. -q -q for less.
      -l logfile                 Use logfile rather than stderr for diagnostics.
      -?                         Display this usage.
    
    avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>

fauxpark, 2020-07-15:
Duplicate of #160

brez1623, 2020-07-15:
> Duplicate of #160

I saw this and figured it would be worth noting the bug still exists. Also do you know how to identify what devices to uninstall in device manager? I found the arduino but wasn't sure what other devices to uninstall.

fauxpark, 2020-07-15:
I've actually been digging into the cause of this today, and it turns out when the Toolbox detects a Caterina device with a PID of 8036 or 8037 (there might be more), it's picking up whatever the Arduino sketch is that appears to come preloaded on Pro Micros. This means the Pro Micro is not in bootloader mode; the correct PID you should be seeing is 0037. So now, I think uninstalling doesn't really do much considering the bootloader device is not part of the composite HID/CDC device above.

I'll make a PR in a bit that should improve bootloader detection for CDC-type devices like Caterina, and be a bit more selective of the PIDs we match on.

brez1623, 2020-07-15:
Do you have any recommendation for anything I could try? Sorry I don't know what a PR is?

fauxpark, 2020-07-15:
Basically you need to reset the Pro Micro into bootloader mode, by shorting the RST and GND pins briefly. Then you should see a new device appear in the Toolbox, but with an ID of 2341:0037 instead of 8037.

PR = Pull Request: https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests

brez1623, 2020-07-15:
The reset worked, but now I get this error.

*** Caterina device connected: Microsoft USB Serial Device (COM7) (2341:0037:0001)
*** Attempting to flash, please don't remove device
>>> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"E:\Downloads\spaceman_2_milk_layout_mine.hex":i -P COM7
    
    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44
    
    avrdude.exe: ser_drain(): read error: The I/O operation has been aborted because of either a thread exit or an application request.
    
    
    avrdude.exe: AVR device initialized and ready to accept instructions
    
    Reading | avrdude.exe: ser_send(): write error: sorry no info avail
    avrdude.exe: ser_recv(): read error: The device does not recognize the command.
    
    
    avrdude.exe: butterfly_recv(): programmer is not responding
    ################################################## | 100% -0.00s
    
    avrdude.exe: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)
    
    Reading | avrdude.exe: ser_send(): write error: sorry no info avail
    avrdude.exe: ser_recv(): read error: The device does not recognize the command.
    
    
    avrdude.exe: butterfly_recv(): programmer is not responding
    ################################################## | 100% -0.00s
    
    avrdude.exe: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)
    
    Reading | avrdude.exe: ser_send(): write error: sorry no info avail
    avrdude.exe: ser_recv(): read error: The device does not recognize the command.
    
    
    avrdude.exe: butterfly_recv(): programmer is not responding
    ################################################## | 100% 0.00s
    
    avrdude.exe: Device signature = 0xffffff (probably .reduced_core_tiny)
    avrdude.exe: Yikes!  Invalid device signature.
                 Double check connections and try again, or use -F to override
                 this check.
    
    avrdude.exe: ser_send(): write error: sorry no info avail
    avrdude.exe: ser_recv(): read error: The device does not recognize the command.
    
    
    avrdude.exe: butterfly_recv(): programmer is not responding
    avrdude.exe: error: programmer did not respond to command: leave prog mode
    avrdude.exe: ser_send(): write error: sorry no info avail
    avrdude.exe: ser_recv(): read error: The device does not recognize the command.
    
    
    avrdude.exe: butterfly_recv(): programmer is not responding
    avrdude.exe: error: programmer did not respond to command: exit bootloader
    
    avrdude.exe done.  Thank you.
    
*** Caterina device disconnected: Microsoft USB Serial Device (COM7) (2341:0037:0001)
*** HID console disconnected (FEED:B195:0001)

brez1623, 2020-07-16:
I was able to figure it out. Thank you for all of the help!

SimonBerens, 2020-07-30:
> I was able to figure it out. Thank you for all of the help!

What did you do to fix it?

closed 2020-08-05 by fauxpark