qmk/qmk_toolbox #210: avrdude: butterfly_recv(): programmer is not responding

semilin, 2020-08-03:
## Describe the Bug

When running qmk toolbox on a mac and writing it to a pro micro, I get an error which is put in to the context section. When running the generated avrdude command by itself, the same issue occurs. Running that same command on Void Linux, again, the error occurs.

## System Information

 - Operating system: Void Linux + Mac OS High Sierra
 - QMK Toolbox version: Latest (compiled from source)

## Additional Context

```
$ sudo avrdude -p atmega32u4 -c avr109 -U flash:w:reviung41_default.hex -P /dev/ttyACM0

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =
    Software Version = .; Hardware Version = .
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
```

nooges, 2020-08-03:
You’ll need to double tap reset/GND to get the Pro Micro to get into the correct bootloader mode, as what you’re currently seeing occurs when you plug the stock Pro Micro in without doing a reset.

semilin, 2020-08-03:
I did that (by wiring a button myself) but now it says that there are no devices available. Did I brick it?

semilin, 2020-08-03:
Ah, I needed to disconnect the button before flashing. Duh. Thank you, it flashed just fine now!

closed 2020-08-03 by semilin