qmk/qmk_toolbox #212: QMK Toolbox automatically resetting my board after flashing

gregford117, 2020-08-05:
hey,

so i trip the reset on my pcb and DFU device is connected.

I attempt to flash my hex file, flash completes successfully but then it repicks up the device as DFU? then the board is unresponsive? it seems the board is auto resetting after each flash?

Would really appreciate if someone could help

screenshot of toolbox output for info

[image]

RowanAldean, 2020-09-02:
Did you ever resolve this? I'm having the same problem for the same MCU and same board

gregford117, 2020-09-14:
@RowanAldean no still having the issue.

I can use TMK but trying to get QMK to work stilll

RowanAldean, 2020-09-14:
> @RowanAldean no still having the issue.
> 
> I can use TMK but trying to get QMK to work stilll

Created a nice TMK guide and will look at hopefully improving the existing tools or allowing QMKbuilder support atleast for the TKG chrome app. If I end up doing that it'll most likely be around xmas time - for now this may help someone/you:

https://www.reddit.com/r/MechanicalKeyboardsUK/comments/iljg5o/flashing_guide_how_to_flash_the_xd84_pro_pcb_uk/?utm_source=share&utm_medium=web2x&context=3

CapsLock LED is self explanatory using the TKG tools too if you need that - I personally don't need any of the QMK features beyond what is in the guide (function layer and ISO) and capslock led so I won't be staying up-to-date with this issue, however hopefully the bug is fixed soon :+1:

fauxpark, 2020-09-14:
Is this the XD84 Pro? If so, no wonder it's not working - it's actually a different board to the XD84 that is already in QMK. 

There's a PR to add support to the Pro here: https://github.com/qmk/qmk_firmware/pull/9750 but it hasn't been merged yet.

RowanAldean, 2020-09-14:
> Is this the XD84 Pro? If so, no wonder it's not working - it's actually a different board to the XD84 that is already in QMK.
> 
> There's a PR to add support for the Pro here: [qmk/qmk_firmware#9750](https://github.com/qmk/qmk_firmware/pull/9750) but it hasn't been merged yet.

Yes for the Pro - I did notice it's a different board and hence gave up. Will take a look at that PR out of interest soon :) Thanks for letting us know it's up

grug7, 2020-09-15:
> Is this the XD84 Pro? If so, no wonder it's not working - it's actually a different board to the XD84 that is already in QMK.
> 
> There's a PR to add support for the Pro here: [qmk/qmk_firmware#9750](https://github.com/qmk/qmk_firmware/pull/9750) but it hasn't been merged yet.

Yeah mine is an XD84 Pro also

Thanks for the info. I've watched the PR

Issue can be closed as its due to using the incorrect PCB in QMK

gregford117, 2020-09-15:
> Yes for the Pro - I did notice it's a different board and hence gave up. Will take a look at that PR out of interest soon :) Thanks for letting us know it's up

Yeah mine is an XD84 Pro too.

I have watched the PR and will close this issue as its cause is using the incorrect PCB

Thanks for the help @fauxpark

closed 2020-09-15 by gregford117