qmk/qmk_toolbox #213: [Bug] Ergodox infinity default compile needs dfu-suffix.exe

GNO21, 2020-08-06:
## Describe the Bug
A CLI compile of an Ergodox Infinity default requires dfu-suffix.exe.  It is not included in the toolbox.   
  
 I'm not sure if this is a bug.  The online tool won't compile an ergodox either, I'm guessing this is why.  

## System Information

 - Operating system: Windows 10/Msys2
 - QMK Toolbox version: 0.0.18

## Additional Context

on a fresh QMK msys2 installation, the command:  
`make ergodox_infinity:default` 
   
results in 
`Linking: .build/ergodox_infinity_default.elf                                                        [OK]
Creating binary load file for flashing: .build/ergodox_infinity_default.bin                         [OK]
sh: line 1: dfu-suffix: command not found
make[1]: *** [tmk_core/rules.mk:292: .build/ergodox_infinity_default.bin] Error 127
`

fauxpark, 2020-08-06:
You don't have dfu-util installed. After compilation, dfu-suffix (which comes with dfu-util) is run against the produced .bin file. It does not need to be included in the Toolbox as it's assumed the .bin file you're trying to flash is complete.

https://github.com/qmk/qmk_firmware/blob/master/tmk_core/rules.mk#L293

GNO21, 2020-08-06:
> You don't have dfu-util installed. After compilation, dfu-suffix (which comes with dfu-util) is run against the produced .bin file. It does not need to be included in the Toolbox as it's assumed the .bin file you're trying to flash is complete. (I believe dfu-util currently will allow you to flash .bin files without a valid suffix, it will just complain about it).
> 
> https://github.com/qmk/qmk_firmware/blob/master/tmk_core/rules.mk#L293

Correct, the full dfu-util wasn't installed.   It was using the QMK-toolbox instead.  That's why I wasn't sure if this was a bug and it was supposed to be included or not.   Installing it separately works.

GNO21, 2020-08-06:
This doesn't look to be a bug as it wasn't intended to be included.

closed 2020-08-06 by GNO21

fauxpark, 2020-08-06:
For context, the Toolbox bundles its own copies of the flashing tools, entirely independent from what you might have installed with Homebrew or MSYS. So you might be able to flash in the Toolbox, but not compile in the terminal.