qmk/qmk_toolbox #222: Massdrop Chip IDs Missing

mcrosson, 2020-09-14:
## Describe the Bug

There are a number of mass drop chip ids missing from the included loader. Would it be possible to have these missing ids added to the build of mdloader used by QMK Toolbox?

## System Information

 - Operating system: All
 - QMK Toolbox version: 0.0.19

## Additional Context

The full list of chip ids I've been able to find (including one from the 2020/09/11 batch of ALTs) is below

```

mcu_t mcus[] = {
      //Name,       Chip ID     Chip ID,    Program Memory, Data Memory,    Program Addr,   Data Addr
      //            Address                 (FLASH_SIZE)    (HSRAM_SIZE)    (FLASH_ADDR)    (HSRAM_ADDR)
    { "SAMD51J18A", 0x41002018, 0x60060006, 0x40000,        0x20000,        0x00000000,     0x20000000},
    { "SAMD51J18A", 0x41002018, 0x60060305, 0x40000,        0x20000,        0x00000000,     0x20000000},
    { "SAMD51J18A", 0x41002018, 0x60060306, 0x40000,        0x20000,        0x00000000,     0x20000000},
    { "SAMD51J18A", 0x41002018, 0x60060005, 0x40000,        0x20000,        0x00000000,     0x20000000},
};

```

tzarc, 2020-09-14:
I'd recommend making a PR against mdloader itself, then once it's merged upstream pinging us to produce an upgraded build.

mcrosson, 2020-09-14:
Looks like more than a few folks have beat me to the main mdloader sources, closing issue since this is not your problem and all on mass drop at this point.

closed 2020-09-14 by mcrosson